felis/USB_Host_Shield_2.0 #313: USBH_Midi bulk flag

vit3k, 2017-09-04:
`if ((epDesc->bmAttributes & 0x02) == 2) {//bulk`

I think this should be changed to:

`if ((epDesc->bmAttributes & 0x03) == 2) {//bulk`

because 0x02 mask will match bulk and interrupt modes and wrong endpoint will be used.

Bits 0..1 Transfer Type
00 = Control
01 = Isochronous
10 = Bulk
11 = Interrupt

Lauszus, 2017-09-04:
Thank you! I can see that bug is actually in several drivers. I'll take a look at it later today.

Lauszus, 2017-09-05:
See: https://github.com/felis/USB_Host_Shield_2.0/pull/314.

closed 2017-09-06 by Lauszus

YuuichiAkagawa, 2017-10-30:
novation Launchpad has illegal endpoint attribute.
This device reports bmAttributes.TransferType is 3. But it is bulk transfer device.
Old implementation is work fine with MIDI.  I will fix later.
```
if ((epDesc->bmAttributes & USB_TRANSFER_TYPE_BULK)) {//bulk
```

vit3k, 2017-10-31:
So how can I then choose correct endpoint when I have both (bulk and interrupt) in my device (Boss Katana 100)?

Lauszus, 2017-10-31:
@vit3k I agree, the code is correct as it is right now, but my guess is that @YuuichiAkagawa will make a special check for that particular device or create a fallback if the current implementation fails to initialise the device.

YuuichiAkagawa, 2017-10-31:
@vit3k Please show me your descriptors using USB_desc.

PR #314 (2017-09-06): Fixed bit mask indicating the transfer type when reading the attribut…
…es in an endpoint

Fixes #313
commit 597d389a (2017-09-06): Fixed bit mask indicating the transfer type when reading the attributes in an endpoint

Fixes #313