felis/USB_Host_Shield_2.0 #320: broken link http://www.circuitsathome.com/products-page/arduino-shields

lzhang10, 2017-10-05:
One of the purchase links in the README: http://www.circuitsathome.com/products-page/arduino-shields, no longer works. This should be corrected or removed.

DSuarez93, 2019-05-02:
As of now and for some time, the circuitsathome website domain seems to be down. If the website will not be restored, its links should be removed from the readme.

eplumer, 2019-05-26:
Actually, no, don't remove it. As is, you can plug the URL into the Way Back Machine (archive.org) and retrieve all of the history of the articles which are massively helpful. If you take the link out, you are dead in the water.

jwyse, 2019-11-30:
It looks like commit cd87628af4a693eeafe1bf04486cf86ba01d29b8 corrects this issue.