felis/USB_Host_Shield_2.0 #321: Using USB Host Shield to connect (via a Bluetooth Dongle) with mini Gamepad bluetooth controler

jkoenig72, 2017-10-07:
Hi,

just wondering. Doing e-skateboard right now. I was thinking it could be a good idea to use those cheap Bluetooth mini game pads, cost around 4$ and very small.

[](https://www.aliexpress.com/item/New-4-in1-Mini-Wireless-Portable-Bluetooth-Remote-Gamepad-Game-Controller-Joystick-For-Samsung-Gear-VR/32815681614.html?spm=2114.search0104.3.1.k865Tb&ws_ab_test=searchweb0_0,searchweb201602_2_10152_10065_10151_10068_10344_10342_10343_10340_10341_10193_10194_10304_10307_10060_10302_10155_10154_10056_10055_10054_5370015_10059_10534_10533_10532_100031_10099_10338_10339_10103_10102_10052_10053_10107_10050_10142_10051_10320_10321_10322_10084_10083_10080_10082_10081_10177_10110_5590015_10111_10112_10113_10114_10180_10312_10313_10314_10184_10319_10078_10079_10073_10186,searchweb201603_30,ppcSwitch_7&btsid=693121df-3886-44cb-9a6d-95ef2a8128b4&algo_expid=301d85b6-9353-407f-a998-7e5a69db0eee-0&algo_pvid=301d85b6-9353-407f-a998-7e5a69db0eee)

Lauszus, 2017-10-08:
@jkoenig72 if it support Bluetooth HID, then it is already supported: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Bluetooth/BTHID/BTHID.ino.

jkoenig72, 2017-10-09:
Thanks. I use this one. Btw. great work!

A few ideas:

I would like to define a fixed Bluetooth address for the controller. Best would be directly in the BTHID.ino.

Also - any example of the gamepad parser? For now a very simple one would do it, based on the keyboard parser but simply pass through all hex values in some array. Ideal I need to include the new class in the BTHID so I can do the Motor control from there. Just parse through all events from the controller. Some events get triggered with the mouse parser 0x01, some with 0x02 and some with 0x03. How I can get them all handled in one new parser, gamepad parser?

Thanks!
Joerg

jkoenig72, 2017-10-09:
When I switch debug info on I can see all events showing up in the log - 

L2CAP Interrupt: A1 02 00 07 07 00 X1: 00	Y1: 07	X2: 07	Y2: 00	Rz: 00

What would help is a really simple parser that allow me to have that data in the level of the main BTHID.ino file

Any way to do so?

/Joerg

jkoenig72, 2017-10-09:
This one here points me to the right direction:

https://blog.tkjelectronics.dk/2013/12/bluetooth-hid-devices-now-supported-by-the-usb-host-library/

Seems that the PS4BT example works mostly OOTB with a 4.0 BT Dongle and all those mini gamepads. Thanks for this work you did - love it!

jkoenig72, 2017-10-09:
Minor issue - it only pairs when I have the PIN in the code.

PS4BT PS4(&Btd, PAIR);

When I use (after successfully paired 1st time)
PS4BT PS4(&Btd);

it shows: 
...
Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:13
Wait For Incoming Connection Request
PS4 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0010
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	01
Bluetooth Dongle Initialized
No response to HCI Reset
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:13
Wait For Incoming Connection Request
Keyboard is connecting
Class of device: 00 05 40
Incoming Connection Request
Remote Name: MOCUTE-036_B53-70E8
Connection established
Connected to Device: 16:07:15:03:70:E8
Wait For Incoming Connection Request
HCI Disconnected from Device
Wait For Incoming Connection Request
Keyboard is connecting
Class of device: 00 05 40
Incoming Connection Request
Remote Name: MOCUTE-036_B53-70E8
Connection established
Connected to Device: 16:07:15:03:70:E8
Received Key Request
Wait For Incoming Connection Request
HCI Disconnected from Device
Wait For Incoming Connection Request

Any idea?

jkoenig72, 2017-10-12:
I can confirm that this is working simply by using the PS4BT example.

Lauszus, 2017-10-12:
@jkoenig72 sorry, but you last comment is a bit unclear. Does it work or not with the PS4BT example? From the previous comment is sounds like it does not work when in "non-paring" mode.

jkoenig72, 2017-10-14:
Sorry - yes, it was unclear: it only works with 

PS4BT PS4(&Btd, PAIR);

When I use (after successfully paired 1st time)

PS4BT PS4(&Btd);

It does not work. However as I plan to use it wioth fixed BT address this is ok for me - I always use the pairing mode. 

If connection get lost its important that the motor slowly goes down in speed and in case of reconnecting I can place a reset button outside the case.

Now I run into another issue - I have mapped all keys of the BT controller nicely to a switch statement in loop() - however when I try to add the servo library it works but with seconds of delay, guess thats to some interrupt conflict.

Any idea how I can control an ESC (for the motor speed control) with that - normally servo lib is good for that and easy to use - you pass values between 800 and 200 as speed into esc.writeMicroseconds(speed);  

But seems that both servo lib and USB host libs does not work well together. 

Any other way of controlling an ESC? 

Regards
Joerg

My current code (ESC commented out) -

/*
 Example sketch for the PS4 Bluetooth library - developed by Kristian Lauszus
 For more information visit my blog: http://blog.tkjelectronics.dk/ or
 send me an e-mail:  kristianl@tkjelectronics.com
 */

#include <PS4BT.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>
#include<Servo.h> 

// Controller BT address
uint8_t bta[6] = {0xE8,0x70,0x03,0x15,0x07,0x16};

USB Usb;
//USBHub Hub1(&Usb); // Some dongles have a hub inside
BTD Btd(&Usb, bta); // You have to create the Bluetooth Dongle instance like so

/* You can create the instance of the PS4BT class in two ways */
// This will start an inquiry and then pair with the PS4 controller - you only have to do this once
// You will need to hold down the PS and Share button at the same time, the PS4 controller will then start to blink rapidly indicating that it is in pairing mode
PS4BT PS4(&Btd, PAIR);

// controller vars

int speed = 800;
int up = 20;
int down = 20;
int max = 2000;
int min = 800;
// Servo esc; 

boolean ready = false;

unsigned long previousTime;
unsigned long elapsedTime;

// functions & procedures

bool printAngle, printTouch;
uint8_t oldL2Value, oldR2Value;

void setup()
{
  
//  esc.attach(10); 
  Serial.begin(115200);

#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif

  if (Usb.Init() == -1) {
    Serial.print(F("\r\nOSC did not start"));
    while (1); // Halt
  }
  Serial.print(F("\r\nPS4 Bluetooth Library Started\r\n"));

  previousTime = millis();
  
}
void loop() 
{
  Usb.Task();

  if (PS4.connected()) 
  {
    elapsedTime =  millis() - previousTime;
    if (Btd.connected!=false & (elapsedTime > 99 ))
    {  

      switch (PS4.getAnalogHat(LeftHatX)) {
      case 127:    // after reboot
        if (ready==false)
        {  
          ready = true;
          Serial.println("after reboot - BEEEP!");
          Serial.print("Speed:");
          Serial.println(speed);
        }
        else
        {
          // not used
        }
        break;
      case 0:    // nothing
        //Serial.println("nothing (speed slowly decrease)");
        if (speed>min)
          speed-=25;
        Serial.print("Speed:");
        Serial.println(speed);
        break;
      case 80:    // up
        //Serial.println("up (speed increase)");
        if (speed<max)
          speed+=100;
        Serial.print("Speed:");
        Serial.println(speed);
        break;
      case 79:    // down
        //Serial.println("down (spped downcrease)");
        if (speed>min)
          speed-=100;
        Serial.print("Speed:");
        Serial.println(speed);
        break;
      case 82:    // right
        Serial.println("right");
        break;
      case 81:    // left
        //Serial.println("left (hold speed)");
        Serial.print("Speed:");
        Serial.println(speed);
        break;
      case 34:    // select
        Serial.println("select");
        break;
      case 39:    // start
        Serial.println("start");
        break;
      case 32:    // X
        Serial.println("X");
        break;
      case 30:    // A
        Serial.println("A");
        break;
      case 33:    // Y
        Serial.println("Y");
        break;
      case 31:    // B
        Serial.println("B");
        // soft_restart();
        break;
      }

      // check ranges
      if (speed > max)
        speed = max;
      if (speed < min)
        speed = min; 

      //delay(100);  
      //esc.writeMicroseconds(speed);
      //delay(100);
      /*  
      Serial.print(previousTime);
      Serial.print(",");
      Serial.print(elapsedTime);
      Serial.print(":");
      Serial.print(" ControllerStatus: ");
      Serial.println(PS4.getAnalogHat(LeftHatX));
      */
      
      previousTime = millis();
      
      // check ranges
      if (speed > max)
        speed = max;
      if (speed < min)
        speed = min;  
    }
  }
}

jkoenig72, 2017-10-14:
Found the information that pins 9 through 13 used by default. Changed to 7 now - works!

hruckertodcom, 2017-12-21:
Hello!
I have a problem, could you help me or share a piece of code?

`no matching function for call to 'BTD::BTD(USB*, uint8_t [6])'`

I understand that you changed the function BTD ()