felis/USB_Host_Shield_2.0 #324: ESP32 support request

kn1cht, 2017-10-29:
Hello,

I want to use USB Host Shield with ESP32(Dev Module), but found this library not support the module.

I personally tried to add ESP32 pin setting to the code, based on ESP8266 related commits.
Although compiling succeeded, it didn't work(`Usb.Init()` failed).

Is there any plan to officially support running USB Host Shield with ESP32 board?

The below are pin settings in my attempt, for your reference.

| USB Host Shield | ESP32 Dev Module | role |
|-----------------|------------------|----------|
| 5V | 5V |  |
| GND | GND |  |
| RESET | EN |  |
| D9 | GPIO17 | INT |
| D10 | GPIO5 | SPI SS |
| D11 | GPIO23 | SPI MOSI |
| D12 | GPIO19 | SPI MISO |
| D13 | GPIO18 | SPI SCK |

Lauszus, 2017-10-31:
I haven't looked into it, but it should be possible. Can you share your code somewhere, so I can see what you have done so far?

kn1cht, 2017-10-31:
@Lauszus Thank you for reply!
Changed code is in my forked repository.
https://github.com/felis/USB_Host_Shield_2.0/compare/master...kn1cht:ESP32?diff=split

Lauszus, 2017-11-01:
@kn1cht I had a quick look and it looks okay to me. Please try the following example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/board_qc/board_qc.ino and post the output.

kn1cht, 2017-11-01:
@Lauszus result of board_qc.

```
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision invalid. Value returned: 00
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test
```

Same output as when shield was not connected. 
So, most likely problem is that I failed with pin number setting, or pins I used is not working correctly.
(I have double checked my circuit.)

Lauszus, 2017-11-04:
Do you have access to an oscilloscope, so you can probe the pins?

kn1cht, 2017-11-05:
@Lauszus Not right now, but I can do it in several days. What kind of tests should I do?

Lauszus, 2017-11-05:
Simply probe the SPI lines and see if you can see communication going back and forth. This would tell us if the SPI is working on not.

kn1cht, 2017-11-06:
SCK and MOSI seem to be working.

[image]

I used Arduino official USB Host Shield, so maybe I had to shift voltage to 5V.

kn1cht, 2017-12-23:
@Lauszus I apologize for long time no updates.
I obtained USB Shield for Arduino Pro Mini, and ran `board_qc` example on ESP32.
It seems to be working without problem!

[image]

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:956
load:0x40078000,len:0
load:0x40078000,len:11856
entry 0x40078a34

Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K................ SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Value written: 00 Value read: FF 
Press any key to continue...
```

I also tryed `hub_demo` and `USB_desc`, and they worked as expected.

xxxajk, 2017-12-23:
@kn1cht  The GPIO test will fail without a testing jig, so the above looks fine. Just hit send on the Serial monitor and it will continue the test.
Since I'm going to be using this in an actual real-world product, I thought I would chime in...
Right now it is under test, and has passed the 24 hour mark as of this writing, which is more than good enough for the target application that my client needs to use it for.

Lauszus, 2018-01-10:
@kn1cht could you send a pull request with your code? :)

ebrahim-Qaashah, 2018-01-13:
Hi all,
  I have just modified the USB host shield 2.0 library for ESP32.
  Visit my repository below:
   https://github.com/ebrahim-Qaashah/USB_Host_Shield_2.0_Support_ESP32

Lauszus, 2018-01-14:
@ebrahim-Qaashah please checkout this pull request: https://github.com/felis/USB_Host_Shield_2.0/pull/340 done by @kn1cht. Great if you could test it and confirm that it is working on your side as well.

closed 2018-01-15 by Lauszus

jakeammons, 2018-02-11:
I followed the connection diagram at the top of this request, but have not yet had success with the quality control routine. I am using a USB Host Shield Mini and an ESP32. Any suggestions?

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision invalid. Value returned: 00
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test`

kn1cht, 2018-02-11:
@jakeammons 
I have to correct one point in my first comment(now updated it).

You should connect `RST` pin of UHS to VCC, not `EN` pin.
So when you use UHS Mini, `RST` should be connected to 3.3V pin.

I can't remember where I got this information, but my UHS worked with this method.

xxxajk, 2018-02-11:
Actually you should connect reset to 2.2k resistor to 3.3v. it will be a
lot more stable that way.

coletz, 2019-04-14:
Hi guys, sorry for reopening but I'm having some issue connecting the ESP32 to UHS mini v2 module. So I just bought this module and discovered that there are (at least) 2 versions: [here you can see official vs chinese](https://www.arduino.cn/thread-81435-1-1.html). I have the "chinese" one. I've seen a lot of people connecting arduinos or teensy to this module, and everyone cuts the connection with the 2.2k resistor in order to provide 5v instead of 3.3v from the raw pin (explained [here](https://youtu.be/5qXv7TJI324?t=58)). Should I do something similar even with the ESP32? Which pin should I connect since there's no raw? any 5v is ok? Thanks!

Lauszus, 2019-04-15:
@coletz in general devices that needs to be powered via USB needs the VBUS to be 5V. Only self-powered devices will work with 3.3V.

Yes you can use any 5V. Just make sure it can supply 500 mA at minimum.

coletz, 2019-04-15:
Ok so after struggling for some time I've connected my esp32 to the uhs mini v2. I was getting @jakeammons error using 5v, but connecting everything to 3v3 fixed the issue. Since this is pretty strange I'm assuming my board is partially broken, but if someone is having problems it could be useful to test everything on 3v3 (without cutting the wire as shown by other).

samuelbles07, 2019-05-09:
Hi, in my UHS, it doesn't have a RST pins, is it MAX3421E reset pin? I'm using this [board](http://shop.tkjelectronics.dk/product_info.php?products_id=43). In the schematics provided in that page, MRST of MAX321E connect to 74AHC125 so i just solder MRST -> 2.2k -> 3.3v like in the photo below 

[image]

When i testing it the result is :

`Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision invalid. Value returned: 00
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test`

In the schematic, no connection to the MAX from arduino pin shield (D9-D13). Is there something i had missed?

dclx, 2019-06-05:
I tried assert reset pin in setup() before access MAX3421E.

pinMode( 7, OUTPUT);  // MAX3421E RESET pin
digitalWrite(7, 1); digitalWrite(7, 0); delay(10); digitalWrite(7, 1); // reset

 ESP32 pass ：D

Mikegyver, 2020-02-10:
@kn1cht what i underderstand when u use arduino usb host shield, the ESP32 can't communicate with the shield.

> 
> 
> I used Arduino official USB Host Shield, so maybe I had to shift voltage to 5V.

but when u use the usb host shield for arduino pro mini, the example code worked as expected.

> 
> 
> @Lauszus I apologize for long time no updates.
> I obtained USB Shield for Arduino Pro Mini, and ran `board_qc` example on ESP32.
> It seems to be working without problem!
> 
> I also tryed `hub_demo` and `USB_desc`, and they worked as expected.

may i know do u have ever try back on normal arduino usb host shield after that?

kn1cht, 2020-02-10:
@Mikegyver No, I haven't tried normal Arduino USB Host Shield.

Mikegyver, 2020-02-10:
@kn1cht may i get opinion regarding this issue?

https://github.com/felis/USB_Host_Shield_2.0/issues/511

unfortunately i don't have with me usb host shiled for arduino pro mini. for the time being i only have the arduino official usb host shield. i already ordered the pro mini usb host shield, but, it kind of drag since the seller is from china - i don't expect i could get it near soon.

i have go through both schematic. both schematic are exactly same. i totally have no idea why it won't work on normal arduino usb host shield. since u are the contributor for this feature, i think u are might the best person to ask opinion. best regards.

kn1cht, 2020-02-10:
@Mikegyver 
> i have go through both schematic. both schematic are exactly same. i totally have no idea why it won't work on normal arduino usb host shield.
The shield that I tried was Arduino USB Host Shield by Arduino.cc.
https://store.arduino.cc/usa/arduino-usb-host-shield

It seems to be no longer sold and is not the same board as one in #511.
So I'm not sure about your problem with your UNO-size host shield.

Does your DOIT ESP32 board output 5V from the VIN pin?

Mikegyver, 2020-02-11:
> 
> 
> @Mikegyver
> 
> Does your DOIT ESP32 board output 5V from the VIN pin?

yes @kn1cht . VIN pin of the DOIT ESP32 is 5V. but, the max 3421e cannot be supply by 5V right? for Vbus of the max 3421e i supply with 3.3V while 5V (VIN ESP32) only go to the VBUS usb.

kn1cht, 2020-02-11:
@Mikegyver Hmm, I have no more idea about the problem...
I hope your pro mini shield arrives sooner and works as expected.

ebrahim-Qaashah, 2020-02-19:
Hello dear,
Thank you so much. Everything is going OK.

Mikegyver, 2020-02-28:
> 
> 
> @Mikegyver Hmm, I have no more idea about the problem...
> I hope your pro mini shield arrives sooner and works as expected.

@kn1cht may i know RST pin u connect to EN or 3.3V?

kn1cht, 2020-02-29:
> @kn1cht may i know RST pin u connect to EN or 3.3V?

RST should be connected to 3.3V or 5V pin (depends on whether voltage your module operated with).

>| USB Host Shield | ESP32 Dev Module | role |
>|-----------------|------------------|----------|
>| 5V(3.3V in mini shield) | 5V(or 3.3V) |  |
>| GND | GND |  |
>| RESET | ~EN~ 5V or 3.3V |  |
>| D9 | GPIO17 | INT |
>| D10 | GPIO5 | SPI SS |
>| D11 | GPIO23 | SPI MOSI |
>| D12 | GPIO19 | SPI MISO |
>| D13 | GPIO18 | SPI SCK |

gomtings, 2021-07-02:
I need to connect esp32 and USB_Host_Shield now, but I can't connect, can I know why?
I'm using ESP32S and I'm using something called USB_Host_Shield.
The library is using USB_Host_Shield_2.0.

[image]

[image]

This is my Shield photo.

[image]
[image]
[image]

This is my ESP32 picture.
I've seen the Wiggle, and I've run the example below.

https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/board_qc/board_qc.ino 

The result is

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision invalid. Value returned: 00
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test

I think ESP32 is all added to the library, but I don't know why not. 

If you look at the writing, Shield needs to revise something. What should I do?
Is it possible that it won't connect?

xxxajk, 2021-07-02:
Not too bad of a clone... Although it is missing the USB power ability.

Your setup doesn't work because it's set for 5V not 3.3v.

```
Remove the AHC and ATC ICs, then bridge the solder jumpers with a dab of solder.
Connect reset to a 1k resistor to 3.3v.

Tips: 
Keep SPI lines as short as possible.
Don't use a breadboard.
```

xxxajk, 2021-07-02:
It also looks like you are thinking of how an UNO onnects, you need to connect to the ICSP connector for SPI, not the uno connections :-)

gomtings, 2021-07-02:
> 또한 UNO가 어떻게 연결되는지 생각하고 있는 것 같은데, UNO 연결부가 아니라 SPI용 ICSP 커넥터에 연결해야 한다 :-)

That's right, Uno connected to the ICSP connector of 6-pin. I saw it and connected it to ESP32.
[image]
[image]

ESP looked at the images above and wired them.

gomtings, 2021-07-02:
> 클론도 나쁘지 않은데... USB 전원 기능이 없음에도 불구하고
> 
> 3.3V가 아닌 5V로 설정되어 있어서 설정이 작동하지 않는 경우.
> 
> ```
> Remove the AHC and ATC ICs, then bridge the solder jumpers with a dab of solder.
> Connect reset to a 1k resistor to 3.3v.
> 
> Tips: 
> Keep SPI lines as short as possible.
> Don't use a breadboard.
> ```

My ESP32 has a 5V output.
So I tried my best to connect with Uno.

"Remove the AHC and ATCIC and then bridge the solder jumper to the solder.
Connect the reset to the 1k resistor at 3.3V."

How shall I do it?
Do you have an example picture of the work?

HisashiKato, 2021-07-27:
@gomtings

[image]

THORZEUS, 2023-07-23:
hi guys!, I couldnt belive that there is not any wiring diagram or picture of esp32 a  with UHS working so, after 3 days debugging and reserching here it is

Note
VBUS TO 5V
RESET TO 3.3v 1K OHM
IC HCT IS FEDEED WITH 3.3V

[image]

THORZEUS, 2023-07-23:
[image]

SaddemHajji, 2024-05-20:
@THORZEUS i need the help please can i talk with please i need help my esp32 devkit1 with usb host shield 2.0

Shuzukochan, 2024-10-30:
> hi guys!, I couldnt belive that there is not any wiring diagram or picture of esp32 a with UHS working so, after 3 days debugging and reserching here it is
> 
> Note VBUS TO 5V RESET TO 3.3v WIRED WITH A 1K OHM IC HCT IS FEDEED WITH 3.3V
> 
> ![ZZ](https://private-user-images.githubusercontent.com/37101659/255378312-974362fe-55b6-4af2-8df3-6a64a2326033.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyODc3ODAsIm5iZiI6MTczMDI4NzQ4MCwicGF0aCI6Ii8zNzEwMTY1OS8yNTUzNzgzMTItOTc0MzYyZmUtNTViNi00YWYyLThkZjMtNmE2NGEyMzI2MDMzLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMwVDExMjQ0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMwOTQ3NzY2OTI0NjM5ZTBkM2NmNjZkMWRlNWQ1NmVlODI2MDc4MTdlNDA0Y2M5MjFjMTQ3NTAzZTUxNzJkZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Ics4_yHjTUEQTxvh7NPcN8HsNSGTdk6LKgHOY3T_Zpk) ![imagen](https://private-user-images.githubusercontent.com/37101659/255378599-cf098631-f0a6-45b1-bb43-6d741add3360.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyODc3ODAsIm5iZiI6MTczMDI4NzQ4MCwicGF0aCI6Ii8zNzEwMTY1OS8yNTUzNzg1OTktY2YwOTg2MzEtZjBhNi00NWIxLWJiNDMtNmQ3NDFhZGQzMzYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMwVDExMjQ0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ2ZTkwNTE5OGJmZDJiNWM1YTQ1NDI1NGQ1ZDNlZTA3Nzg4YjQxMTNiY2U0Y2Y4Njg3NDMwMGEyYmE4NWI2MGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9kLooCxwNI-MVWOWo1yqUB4A1yxzLCM1fHGDaPKJROs)
> 
> Gracias totales!

Can I have your diagram? I don't understand in IC HCT IS FEDEED WITH 3.3V.
i have both usb host shield 2.0 and usb host shield mini, both do not work with ESP, same error in this post

PR #341 (2018-01-15): Added support for ESP32
Supersedes #340 and fixes #324
PR #340 (2018-01-15): Add ESP32 pin settings 
fix #324 
Supported Espressif ESP32.