felis/USB_Host_Shield_2.0 #328: usb host on stm32L4 Dragonfly support

arpeggio1234, 2017-11-19:
I am trying to compile an Arduino usb hostshield for the STM32L4 based Dragonfly code and the compiler says ''please define board in avrpins.h''.
How can i do this, i am on Linux, manjaro, and found the avrpins.h file and see there is no 
STM32L4 entry, can i put it in, and yes what? Or can i use the teensy pinout for the dragonfly since they are lookalikes.

Thanks in advance, Dian from the Netherlands

arpeggio1234, 2017-11-29:
> Please checkout the pull request I did for the STM32F446 Nucleo board: https://github.com/felis/USB_Host_Shield_2.0/pull/178.
>
> A PR is more than welcome!
>
> https://github.com/felis/USB_Host_Shield_2.0/issues/328#issuecomment-347224258

Hello Kristian, thanks for making all this great stuff.For now just trying to send you a pm, more to come soon. Am trying to put usb shield on Dragonfly.
Dian from the Netherlands

Lauszus, 2017-11-29:
You can find my contact details here on Github: https://github.com/Lauszus, but I prefer to keep the conversation here as an issue, so other can find it in the future.

arpeggio1234, 2017-11-29:
Oh sorry you meant a pull request.
I was 'reading' a pm.
A pull request for the STM32L4 is a good idea. What csn i do to ask that, make it happen,  can i br of any help.
Many thanks again for yours concern.
Dian

Lauszus, 2017-12-05:
@arpeggio1234 simply fork the repository and then start modifying the code. Then once you are done you can simply open up a pull request.

Sorry, but I have limited time at the moment to help you out. I would recommend you open up a pull request and then I'll help you a long the way.

Lauszus, 2018-03-15:
I will close this issue, as I believe it has been solved.

closed 2018-03-15 by Lauszus