felis/USB_Host_Shield_2.0 #330: Will CD/DVD drive work with USB-to-SATA cable?

jsweet13, 2017-11-30:
Hello everyone, 

I was wondering if we can send atapi packets or mmc commands to SATA drives with this library?

I haven't really worked with USB as it is pretty difficult to understand for me, but I was wondering if anyone has been able to read or write 2048 byte sectors to or from optical media?

Thanks

xxxajk, 2017-11-30:
Short answer: Yes, you can, but you will need an MCU with large amount of RAM, and UHS3 is actually better suited to the task as well. You are also better off using SCSI commands too. Nobody uses ATAPI/MMC over USB.

I've used DVD-RAM on UHS3 with 2K sectors on ARM with 64K RAM, but you need to cache FAT data and write it only when you close/commit the DVD for FAT FS or you burn out the media too fast. 

For regular DVD-RAM and CD-RW, you would need to implement UFS in order to prevent media burnout, or implement your own incompatible FS.

For regular DVD/CD, for data you would need to implement ISO9660 and all of the extensions. This too requires a lot of RAM and would also require more flash.

jsweet13, 2017-11-30:
Great, thanks for explaining to where I could understand. I believe we can format the media to make all LBA's accessible for read/write, but they have everything listed in a packet format in the mmc-5 PDF.

Will do a search on scsi command tutorials.

A perfect scenario would be having the ability to read or write one LBA sector at a time. Would the SCSI Format command format the media automatically for us to grant access to all LBA locations? 

Formatting the media with the SCSI command would make the media available for RAW access, correct?

If so, this would be prefect for creating a custom file system. I was actually drafting out the byte layout format for raw binary mode without a traditional table of contents.

Thank you for helping.

jsweet13, 2017-11-30:
This is pretty exciting, we can probably even build a FAT, FAT32 or NTFS read only memory file system on CD, DVD or BD-R media. I'll be soldering up my USB Host Shield, try to format blank media and then try to do some LBA sector reads before trying to write to the LBA's.

xxxajk, 2017-11-30:
Try UHS3, it works MUCH better for this sort of thing and will "just work" with DVD-RAM.
[https://github.com/felis/UHS30](url) is where to get it. 
I'll be adding a more-or-less intelligent cache for the MCU with large enough RAM probabbly this weekend.
The cache will require an MCU with at least 64K SRAM in order to be useful at all on 2K sectors. More is always better, as it will allow a larger cache, for example an MCU with 128K SRAM would allow a nice large 64K cache.

If I remember how I did this in the past...
32K cache yields about 14 2K FAT table sectors.
64K cache yields about 30 2K FAT table sectors.
It really depended on how you were accessing the data.
Briefly, lower sectors got delayed writing until you did a sync on the cache, unless too much pressure required the one with the lowest amount of activity to be written out. Since FAT tables are at the beginning of the media, lowest number sectors get cached without having to know about the file system.

Notes for DVD-RAM, which I formatted under Linux (UHS3 can also format!)

1. Format as FAT32
2. Format with ONE FAT TABLE
3. Format as 2K sectors
4. Format 2K clusters if possible.

jsweet13, 2017-12-01:
OK, will try the UHS3.

Is there a difference between DVD-RAM and a DVD-R?

Maybe the DVD-RAM allows access to any LBA without a need for format while DVD-R would need to be formatted before allowing random access?

This is a very interesting project.

I believe a good technique would be to format optical media with 2048 byte sectors.

Perhaps implementing a 24-bit LBA with three bytes and figure a way to send a Buffer(2048) to the selected LBA, or fill a Buffer(2048) from the selected LBA would be awesome.

I just bought a Bluray writer and will be curious to try reading and writing LBA sectors one at a time.

The LBA scheme seems to be compatible with the DVD LBA layout. 

SCSI commands should work with getting capacity of the media, free space left and total amount of sectors. The commands working with DVD should be compatible with BD-R and maybe BDXL.

Will the UHS3 be OK to use with the USB Host Shield that has MAX3421E?

Thanks!

jsweet13, 2017-12-01:
Come to think of it, BD-R is a bit more complicating as it groups 32, 2,048 byte sectors together in each partition according to this guide: http://www.hughsnews.ca/faqs/authoritative-blu-ray-disc-bd-faq/9-disc-capacity

Though they state we can read a 2048 byte sector back at minimal, perhaps there's a way to write the same way instead of writing 32 sectors at a time. Maybe there's a way to perform a full disc format to support that layout.

xxxajk, 2017-12-01:
Yes, there is a huge difference. 
If you are just interested in reading DVD, you will need to implement ISO9660 filesystem.
DVD-RAM discs, once formatted, can have any filesystem you want on them, although this may confuse some operating systems.

jsweet13, 2017-12-02:
Sounds good. For UDF format, maybe there could be a way to open a session after format and then create a single file that is as big as the disc capacity is, but leave the session open until it is closed.

Could probably open and close the single file as needed. Perhaps in the single file, the layout for a filesystem could be written. Maybe it is possible to have a file without a file name or extension.

Or make the disc bootable in the lead-in to jump to the data section of the disc to treat the single file as an executable binary, maybe the lead-in could be an x86/64 bootloader like GRUB which might see the data area as a partition.

jsweet13, 2017-12-02:
Maybe the filename could be an .iso and have something like isolinux or grub4dos load and run the iso. So the iso could be created on disc in real time, or written to the disc later with a bootloader in the lead-in area of the image that loads the single file.

I suppose anything is possible.