felis/USB_Host_Shield_2.0 #331: Bluetooth Dongle Init Failed, error code: D1, running PS3BT

EPiress, 2017-12-07:
Hi,

So, i have a Bellkin BT dongle, but when I try to run the PS3BT.ino, i get the following error in the serial monitor:

PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Unknown Device Connected - VID: 050D PID: 065A
BTD Init Failed, error code: D1

I have seen a similar issue before, but my arduino does not pair with my android phone, I don't know what else to do!

Thanks in advance,
EPiress

Lauszus, 2017-12-07:
Please run the following example: https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/USB_desc.

EPiress, 2017-12-07:
I get this: 

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	01
Device Protocol:	01
Max.packet size:	40
Vendor  ID:		050D
Product ID:		065A
Revision ID:		0112
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		00DA
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		03
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0010
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Endpoint descriptor:
Endpoint address:	02
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			02
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			03
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			04
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			05
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		FF
Intf. Protocol:		FF
Intf.string:		00

Endpoint descriptor:
Endpoint address:	84
Attr.:			02
Max.pkt size:		0020
Polling interval:	01

Endpoint descriptor:
Endpoint address:	04
Attr.:			02
Max.pkt size:		0020
Polling interval:	01

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		00
Intf. Class:		FE
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		21
Contents:	05881340001001B300

Addr:1(0.0.1)

Lauszus, 2017-12-07:
It looks like it uses a non-standard endpoint data structure, as the bulk and interrupt endpoints are on different interfaces.

Try to replace this line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/BTD.cpp#L232 with:
```C++
ConfigDescParser<USB_CLASS_VENDOR_SPECIFIC, WI_SUBCLASS_RF, WI_PROTOCOL_BT, CP_MASK_COMPARE_ALL> confDescrParser(this);
```

Please also enable extra debugging by uncommenting this line: https://github.com/felis/USB_Host_Shield_2.0/blob/master/BTD.cpp#L20.

EPiress, 2017-12-07:
After doing such modifications I was finally able to connect my BT dongle to my PS3 controller!

Thank you so much!!

Can you explain me your though process? I mean, how did you kno what to look for?

Also, now my serial monitor is on an infinite loop showing the following lines:

00 00 00 00 00 80 80 7E 81 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FD 01 EB 01 8B 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FD 01 EB 01 8C 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FD 01 EA 01 8C 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FC 01 EA 01 8C 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FD 01 EC 01 8C 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FC 01 EA 01 8C 00 06 
00 00 00 00 00 80 80 7E 81 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FC 01 EA 01 8C 00 06 
00 00 00 00 00 81 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FE 01 EC 01 8C 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FD 01 EA 01 8B 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FD 01 EA 01 8B 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FE 01 EB 01 8B 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FD 01 EA 01 8B 00 06 
00 00 00 00 00 80 80 7E 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 05 16 FF D2 00 00 33 1F 77 01 C0 01 FD 01 EA 01 8B 00 06 

Can you explain me why?

Thanks again, 
EPiress

EPiress, 2017-12-07:
I got it, never mind! I had just uncommented a few lines that should be commented!

Thanks a lot Lauszus, you were very helpful!

Cheers,
EPiress

Lauszus, 2017-12-08:
@EPiress I could tell from the device class that it was set to vendor specific (0xFF). 

Can you give me some more information about your dongle, then I will add a fix to the library.

EPiress, 2017-12-08:
Sure! 

The dongle I used is a Belkin Mini Bluetooth 4.0 Adapter - F8T065bf !
Thanks again,
EPiress

Lauszus, 2017-12-10:
@EPiress could you please confirm that #332 solves this issue?

EPiress, 2017-12-10:
Yes, #332 solves this issue!

Lauszus, 2017-12-10:
@EPiress thanks, it is now merged into master :)

commit a06d173a (2017-12-09): Workaround issue with Belkin F8T065bf Bluetooth dongle

Fixes #331
PR #332 (2017-12-10): Workaround issue with Belkin F8T065bf Bluetooth dongle
Fixes #331
closed 2017-12-10 by Lauszus