felis/USB_Host_Shield_2.0 #334: The HID device is not recognized through the USB hub.

XEROHN, 2017-12-13:
Hello!

I want to get the data of the keyboard and mouse through usb hub.
But I have a little problem.

I'm using the USBHIDBootKbdAndMouse.ino file.
`USBHub     Hub(&Usb);`
Of course I wrote the USBHub object.

The environment is the same as the picture.

[image]

connect a mouse or keyboard device (HID device) directly to the Shield USB port, data will be displayed.
But, if through the hub, only devices other than HID devices are recognized.

The serial output data is as follows.

```
Start
BM Init
Addr:0A
NC:01
HID_PROTOCOL_KEYBOARD AND MOUSE
bNumEP:01
D1
BM Init

getDevDescr 0D
BM Init

getDevDescr 0D
BM Init

getDevDescr 0D
BM Init

getDevDescr 0D

Lauszus, 2017-12-13:
@XEROHN from the picture it looks like you are using an unpowered USB hub. Can you try to connect an external power supply to the Arduino?

If that does not work, then I would recommend using a powered USB hub or connecting a 5V power supply directly to the Arduino, as all those USB device might draw a lot of current.

xxxajk, 2017-12-13:
Also, please note that this driver is for keyboard and mouse on a single device, such as a keyboard with built-in trackpad, or a usb to ps/2 convertor. 
For individual keyboard and mouse at the same time, you need to use the two separate drivers.

XEROHN, 2017-12-14:
@Lauszus 
I have tested with a total of six cases, Unfortunately, the output on the serial monitor did not change.

Jumperpad Soldered / Non-soldered
Powered Arduino / Usb hub / Arduino and Usb hub both

Behind the Usb hub shield is a jumper pad.
I've seen the circuit schematic and it seems to affect the power supply.
(On the schematic, the jumper pads(SJ1) are located near the Power pin socket and J2 pin socket.)
[https://www.arduino.cc/en/uploads/Main/USBHostShield-Rev2.pdf](url)

@xxxajk
As far as I know, the HID output data is parsing the data when the event occurs.
So, what do you think of running two Usb.Task in a threads?

Lauszus, 2017-12-14:
@XEROHN jumper SJ1 is just used for the IOREF pin on the Arduino and should not have any effect if you have a newer Arduino with that pin. If that pin is not on your Arduino, then you need to close SJ1.

If that does not work, could you try another hub?

XEROHN, 2017-12-14:
@Lauszus Thank you for your reply.

I have brought out another hub.
When USBHIDBootKbd and USBHIDBootMouse are operated respectively, HID data is outputted well.
But, when USBHIDBootKbdAndMouse.ino was executed, only the address was output and there was no response.

```
Start
BM Init
Addr:0D
NC:01
HID_PROTOCOL_KEYBOARD AND MOUSE
bNumEP:02
D1
BM Init

getDevDescr 0D
BM Init

getDevDescr 0D
BM Init

getDevDescr 0D
BM Init

getDevDescr 0D

Lauszus, 2017-12-14:
It sounds like it is a proven with the combined mouse and keyboard code. Can you try another mouse and/or keyboard?

XEROHN, 2017-12-15:
@Lauszus 

I connected another mouse to the usb hub, but unfortunately the result was the same.

It looks like only the HIDComposite object is created and the Keyboard and Mouse objects of HIDBoot are not created.

USBHIDBootKbdAndMouse.ino
```
USB     Usb;
USBHub     Hub(&Usb);
HIDBoot < USB_HID_PROTOCOL_KEYBOARD | USB_HID_PROTOCOL_MOUSE > HidComposite(&Usb);
HIDBoot<USB_HID_PROTOCOL_KEYBOARD>    HidKeyboard(&Usb);
HIDBoot<USB_HID_PROTOCOL_MOUSE>    HidMouse(&Usb);

KbdRptParser KbdPrs;
MouseRptParser MousePrs;

void setup()
{
  Serial.begin( 115200 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
    Serial.println("OSC did not start.");

  delay( 200 );

  HidComposite.SetReportParser(0, &KbdPrs);
  HidComposite.SetReportParser(1, &MousePrs);
  HidKeyboard.SetReportParser(0, &KbdPrs);
  HidMouse.SetReportParser(0, &MousePrs);
}

void loop()
{
  Usb.Task();
}
```

Is there anything I need to fix in the header file?
This is the only one I have modified.
`#define ENABLE_UHS_DEBUGGING 1`

xxxajk, 2018-10-11:
Will test this later today. Thank you!

apt1987, 2019-03-09:
MOUSE + KEYBOARD IS WORK !!!!! DO LIKE THIS !!!

#include <hidboot.h>
#include <usbhub.h>

// Satisfy IDE, which only needs to see the include statment in the ino.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

void show(uint8_t i)
{
 static uint8_t state = 0;
 state = !state;
 digitalWrite(13, state);
 delay(30);  
}

class MouseRptParser : public MouseReportParser
{
  protected:
    void OnMouseMove(MOUSEINFO *mi);
    void OnLeftButtonUp(MOUSEINFO *mi);
    void OnLeftButtonDown(MOUSEINFO *mi);
    void OnRightButtonUp(MOUSEINFO *mi);
    void OnRightButtonDown(MOUSEINFO *mi);
    void OnMiddleButtonUp(MOUSEINFO *mi);
    void OnMiddleButtonDown(MOUSEINFO *mi);
};
void MouseRptParser::OnMouseMove(MOUSEINFO *mi)
{
  Serial.print("dx=");
  Serial.print(mi->dX, DEC);
  Serial.print(" dy=");
  Serial.println(mi->dY, DEC);
  show(3);
};
void MouseRptParser::OnLeftButtonUp	(MOUSEINFO *mi)
{
  Serial.println("L Butt Up");
};
void MouseRptParser::OnLeftButtonDown	(MOUSEINFO *mi)
{
  Serial.println("L Butt Dn");
};
void MouseRptParser::OnRightButtonUp	(MOUSEINFO *mi)
{
  Serial.println("R Butt Up");
};
void MouseRptParser::OnRightButtonDown	(MOUSEINFO *mi)
{
  Serial.println("R Butt Dn");
};
void MouseRptParser::OnMiddleButtonUp	(MOUSEINFO *mi)
{
  Serial.println("M Butt Up");
};
void MouseRptParser::OnMiddleButtonDown	(MOUSEINFO *mi)
{
  Serial.println("M Butt Dn");
};

class KbdRptParser : public KeyboardReportParser
{
    void PrintKey(uint8_t mod, uint8_t key);

  protected:
    void OnControlKeysChanged(uint8_t before, uint8_t after);
    void OnKeyDown	(uint8_t mod, uint8_t key);
    void OnKeyUp	(uint8_t mod, uint8_t key);
    void OnKeyPressed(uint8_t key);
};

void KbdRptParser::PrintKey(uint8_t m, uint8_t key)
{
  MODIFIERKEYS mod;
  *((uint8_t*)&mod) = m;
  Serial.print((mod.bmLeftCtrl   == 1) ? "C" : " ");
  Serial.print((mod.bmLeftShift  == 1) ? "S" : " ");
  Serial.print((mod.bmLeftAlt    == 1) ? "A" : " ");
  Serial.print((mod.bmLeftGUI    == 1) ? "G" : " ");

  Serial.print(" >");
  PrintHex<uint8_t>(key, 0x80);
  Serial.print("< ");

  Serial.print((mod.bmRightCtrl   == 1) ? "C" : " ");
  Serial.print((mod.bmRightShift  == 1) ? "S" : " ");
  Serial.print((mod.bmRightAlt    == 1) ? "A" : " ");
  Serial.println((mod.bmRightGUI    == 1) ? "G" : " ");
};

void KbdRptParser::OnKeyDown(uint8_t mod, uint8_t key)
{
  Serial.print("DN ");
  PrintKey(mod, key);
  show(1);
  uint8_t c = OemToAscii(mod, key);

  if (c)
    OnKeyPressed(c);
}

void KbdRptParser::OnControlKeysChanged(uint8_t before, uint8_t after) {

  MODIFIERKEYS beforeMod;
  *((uint8_t*)&beforeMod) = before;

  MODIFIERKEYS afterMod;
  *((uint8_t*)&afterMod) = after;

  if (beforeMod.bmLeftCtrl != afterMod.bmLeftCtrl) {
    Serial.println("LeftCtrl changed");
  }
  if (beforeMod.bmLeftShift != afterMod.bmLeftShift) {
    Serial.println("LeftShift changed");
  }
  if (beforeMod.bmLeftAlt != afterMod.bmLeftAlt) {
    Serial.println("LeftAlt changed");
  }
  if (beforeMod.bmLeftGUI != afterMod.bmLeftGUI) {
    Serial.println("LeftGUI changed");
  }

  if (beforeMod.bmRightCtrl != afterMod.bmRightCtrl) {
    Serial.println("RightCtrl changed");
  }
  if (beforeMod.bmRightShift != afterMod.bmRightShift) {
    Serial.println("RightShift changed");
  }
  if (beforeMod.bmRightAlt != afterMod.bmRightAlt) {
    Serial.println("RightAlt changed");
  }
  if (beforeMod.bmRightGUI != afterMod.bmRightGUI) {
    Serial.println("RightGUI changed");
  }

}

void KbdRptParser::OnKeyUp(uint8_t mod, uint8_t key)
{
  Serial.print("UP ");
  PrintKey(mod, key);
  show(2);
}

void KbdRptParser::OnKeyPressed(uint8_t key)
{
  Serial.print("ASCII: ");
  Serial.println((char)key);
};

USB     UsbK;
USBHub     HubK(&UsbK);

HIDBoot < 2 > HubD1(&UsbK);
HIDBoot < 1 > HubD0(&UsbK);

KbdRptParser KbdPrs;
MouseRptParser MousePrs;
void setup()
{
  pinMode(13, OUTPUT);
  digitalWrite(13, 1);
  for(int i = 40; i > 20; i--)
  {
   digitalWrite(13, 1);
   delay(i);
   digitalWrite(13, 0);
   delay(i);
  }
  
  Serial.begin( 115200 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (UsbK.Init() == -1)
    Serial.println("OSC did not start.K");

  delay( 200 );
  
  //Hub.PrintHubStatus();
  
  Serial.print(HubD0.SetReportParser(0, &KbdPrs));
  Serial.print(HubD1.SetReportParser(0, &MousePrs));
  
  //Serial.print(HidKeyboard.SetReportParser(0, &KbdPrs));
  //Serial.print(HidMouse.SetReportParser(0, &MousePrs));
}

void loop()
{
  UsbK.Task();
}

apt1987, 2019-03-09:
[image]
[image]

sudha100, 2019-05-15:
i want a code when mouse is removed then it display mouse removed. if i insert mouse then it display the mouse inserted..............please help me

mattherdelma, 2024-04-02:
When I plug in the mouse and keyboard on the USB hub, the mouse data cannot be read. As a result of multiple tests, it was found that placing the HIDBoot of the mouse in front of the keyboard can read the output of both normally.

`HIDBoot<USB_HID_PROTOCOL_MOUSE> HidMouse(&Usb);
HIDBoot<USB_HID_PROTOCOL_KEYBOARD> HidKeyboard(&Usb);

MouseRptParser MousePrs;
KbdRptParser KbdPrs;`