felis/USB_Host_Shield_2.0 #335: Arduino Uno crashes using XBOXUSB sample

viados, 2017-12-13:
Hey Team!

As the title already says: I'm not able to let run the XBOXUSB Sample using V1.3.1 without continues crashing and restarting my Arduino Uno. As long as I'm pressing the buttons the samples does work. But as soon I'm playing a little more around using the sticks the Arduino will crash.

Are there any known issues or workaround, couldn't find something helpful?
Thanks in advance!

Lauszus, 2017-12-14:
@viados have you modified the code in any way? Please try to connect an external power supply.

Lauszus, 2018-01-20:
Closing due to lack of feedback.

closed 2018-01-20 by Lauszus

viados, 2018-02-10:
Sorry for the late response, I was very busy in the meantime and was able to solve the issue by myself. It seems that there is general issue when creating a new USB class instance using "new". As long as the variable is not on the heap USB shield can be used as expected and demonstrated. I could also not find any other reason for that issue - which also solved my problem.

xxxajk, 2018-02-10:
This is actually a known issue with UHS2, and does not exist in UHS3. Once
UHS3 gets HID, and drivers ported, you will be able to use new if you like.
In fact it is the preferred way with the new library.