felis/USB_Host_Shield_2.0 #337: Can't reconnect PS4usb without resetting Arduino

jordan-edX, 2018-01-08:
Thanks for all the work on this!

I am having an issue in USB mode, which I will try to describe:

I have uploaded the “PS4USB” example sketch from the “USB_Host_Shield_2.0-master” library, unaltered, to an Arduino Mega 2560 board. A Circuits at Home rev 2.0.1 USB host shield is used.

If I plug in a Sony brand PS4 controller with a USB cord, everything works as it should, the serial monitor shows appropriate messages for button clicks and joystick values and the like, and the lights and rumble change on the actual controller correctly.

Now, if I unplug the controller, and then plug it back in without resetting the Arduino, it will not reconnect. If I reset the Arduino, either by power cycling, pushing the physical reset button, or closing and re-opening the serial monitor, the PS4 controller will reconnect.

When I upload the “PS4BT” example sketch and use an appropriate dongle, everything works great and I can pair the controller by holding SHARE and the PS button as outlined in the code, and then disconnect and reconnect the PS4 controller as many times as I want, using the PS button .

The question is, is there a way to make the controller re-connect after unplugging the USB cord, without resetting the system?

My end goal is to set up the system so that it will work in either BT or USB mode. It would be quite nice if everything could be hot swapped so that I don’t need to reset the system if, say, the PS4 controller starts running out of battery, I can just plug in a USB cable and carry on. I have already done this with a PS3 controller using your library, and it works great.

tl:dr is it possible to unplug and replug a USB cable and have it connect every time without resetting the Arduino? Thanks!

Lauszus, 2018-01-09:
You can add a VBUS power switch to the shield. Please read the following blog post: https://www.circuitsathome.com/mcu/vbus-power-control-on-usb-host-shield/.

closed 2018-01-09 by Lauszus

jordan-edX, 2018-01-11:
Thanks for looking at this. I'm still a little confused how that will help with the issue I'm having, my thought process:

From what I can tell looking at the blog post you linked, the VBUS power switch disconnects the 5 volt line to any device plugged into the USB host shield, and also rapidly discharges any capacitors connected to the VBUS line (C3 and C4 on the UHS_20.sch schematic).

My problem happens when the USB cord connecting the PS4 controller to the USB host shield is unplugged. Once that cord is unplugged, anything that happens on the host shield won't be "seen" by the PS4 controller, so I don't understand how turning the VBUS line down will help.

Am I thinking about this wrong? How does the PS4 controller "know" that the VBUS line was pulled low if it is not connected to it? 

If the PS4 controller must be power cycled after it is accidentally unplugged, how would that look? Lets say the USB cord is pulled out, and PS4.connected() returns false, would one just periodically turn on and off the VBUS switch, attempting to catch the controller in a power cycle whenever the human on the other end happens to plug it in?

The controller always connects right away when I reset the Arduino, which I'm pretty sure doesn't pull the VBUS line down hard. Is there no way in software to let the host shield know to attempt to reconnect with the PS4 controller?

Thanks again.