felis/USB_Host_Shield_2.0 #338: USB_HOST_SHIELD on Teensy 3.6

arpeggio1234, 2018-01-08:
Hello, i use the Sparkfun USBhostshield board Rev 3, and it works great with an Arduino Duemilanova.
Use the PS4 controller with it, thanks for making this.
Now i am trying to interface the board and the UHS2 library with the Teensy 3.6. but i tryed all the pin-connecting combinations(not the good one, hahaha), and tryed without the spi4teensy3 option and tryed with SPI1 set Mosi Miso CS Sck in the Setup, but nothing seems to get the board_pc working, i every time get the    Die revision invalid: value retund: FF   error,
Can someone help me out here, have some tips and tricks to know what's happening.
Any ideas how to find out the good pins.
Thanks in advance, Dian from the Netherlands

xxxajk, 2018-01-08:
For teensy 3.6, the on-board host is a work in progress in UHS3.0. If you want to use the UHS board instead, you need to use the mini version, and change the VBUS on it to 5V.

arpeggio1234, 2018-01-08:
Okay thanks man i will think about purchasing another board.
Where can i find how to connect it and finding out that it sure is working,
And just qurious why not the board i have, why is it not working, 
It works with the other Arduino, and don't get me wrong, i will support you.
But i have to make sure what i get, the mini board, that it's working, with the Teensy 3.6.
Thanks again, i know, cause i code alot myself, that's a great job you all are making, the UHS-lib.
Thanks Dian

arpeggio1234, 2018-01-08:
Btw is the shop in Denmark, i am in the Netherlands, so saves my shippingcosts if you where in the states.
Dian

arpeggio1234, 2018-01-08:
And will the full UHS-board work too, with the Teensy 3.6.
Dian

xxxajk, 2018-01-08:
http://shop.tkjelectronics.dk/product_info.php?cPath=22&products_id=45

xxxajk, 2018-01-08:
https://www.circuitsathome.com/mcu/teensy-3-0-now-supported-by-the-usb-host-library/
connections are same on the 3.6 :-)

arpeggio1234, 2018-01-10:
Thanks for the info, i have the usb host shield from sparkfun (rev 3) working now.
Got some bad pin connect issues so i thought it was not working with the standard pin configuration. 
But if i decide to have another board, will the full version of your UHS be an option and power the board  from the teensy3.6 power line from the usb power in on the teensy. This works with the other board to.
I maybe will have the full version cause it has 5 volt and i have an Arduino mega too.
I like these boards.
Dian

arpeggio1234, 2018-01-11:
I ordered the usb mini host shield.
I let you know how i works.
 I like moving around a ps4 controller and grabbing and using that data, for example the control a speakjet or control midi.
Thank for making great ussble gear.
Dian fron the Netherlands

xxxajk, 2018-01-11:
Yes, please follow up. 
Remember that you will have to do a modification to the board for +5 on the USB connector.
The default is 3.3v because the original design was for DSLR cameras (I think).

arpeggio1234, 2018-01-27:
Hello i just got the mini usb host shield. Got it quick working with a mega. But with the teensy 3.6 i got it one time working and later on i tryrd it again and yhen it did not work with the bos

arpeggio1234, 2018-01-27:
Following........
did not work with the board_qc sketch.
The teensy seems to react strange on the UHS, and maybe on it's libraries.
And the combination with it's own usb host libraries. Do you have any ideas about this and tips for finding a good way to get it working on the teensy 3.6.
And by the way the board is good and the pins also, checked it with the mega.
Tganks in advance, Dian

xxxajk, 2018-01-28:
This will be solved at some point in UHS30 (Supporting both at the same time)
Try here https://github.com/felis/UHS30
This will work fine with the board_qc available there

arpeggio1234, 2018-01-28:
Yeah thanks, i tryed that and indeed it worked, the board_qc sketch. But does the UHS 3.0 allready have PS4-controller support. I like that thing.
Kind of flying though Music.
Thank for making this great thing possible and the support. 
Dian

Lauszus, 2018-01-28:
@arpeggio1234 if you have connected a mega to the mini shield directly, then you might have damaged it, as it is __NOT__ 5V tolerant...

arpeggio1234, 2018-01-28:
Oh i connected it to 3.3 volt. But indeed the spi ect, other pins are 5 volt.
I will check it with the Teensy board and the UHS 3.0 lib and the board_qc.
Thanks for the advice. Dian

arpeggio1234, 2018-01-28:
I updated the teensyduino software and lowered the cpu speed of the teensy 3.6, adviced by paul stoffregen. And......it worked with the uhs mini shield. Great.
Thanks for the support and advice.
Dian

Lauszus, 2018-01-28:
@arpeggio1234 great to hear that. Can you determine the maximum frequency where it still works, so I can put it in the documentation :)

Lauszus, 2018-01-28:
@arpeggio1234 also can you check if setting ```USE_SPI4TEENSY3``` to 0 zero: https://github.com/felis/USB_Host_Shield_2.0/blob/master/settings.h#L79 has any effect?

arpeggio1234, 2018-01-31:
Hello Lauszus, i managed to get the UHS mini and teensy 3.6 working after the teensyduino update, with the UHS 2.0 lib, great.
CPU-speed is max, 180 mhz, with or without the spi4 teensy3  setting.
Thanks and let me know if i can do some more tests. Does the UHS 3.0 lib have ps4 support.
Dian

arpeggio1234, 2018-01-31:
And.......above was on my linux manjaro install with arduino ide 1.85 and teensyduino 1.41
On win7 arduino ide 1.85 and teensyduino 1.41 also with UHS 2.0
and on that the board_qc did not work and when i changed the spi4teensy3 to 0 in settings.h it worked.
Great
Dian

Lauszus, 2018-03-15:
@arpeggio1234 thanks for the feedback and no the UHS 3.0 lib does not support PS4 yet...

closed 2018-03-15 by Lauszus