felis/USB_Host_Shield_2.0 #339: how can i use PS3.Reset() function in my arduino to reset the PS3 controller ?

Utsav13, 2018-01-09:
i am displaying my main code over here to perform motion on my robot 
but my ps3 stucks sometimes 
so,i think to continuous reset my ps3 controller
but i can not be use because it is protected...

so, how i can stop to stuck my ps3

void loop() {
  Usb.Task();
    if (PS3.PS3Connected)
    {
        // task performance of my robot
    }
}// main loop

Lauszus, 2018-01-09:
The ```PST.Reset()``` function is used to reset the different variables used by the class. You can call ```PS3.disconnect()``` if you want to set everything back to the default values.

But why do you want to do this in the first place? It is better to investigate why the controller gets "stuck". Please share your code and I will have a quick look.

Utsav13, 2018-01-10:
i can not use PS3.Reset()  directly
error  message : - 
Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\Utsav\Desktop\New manual\11 - jan - 2018\code with old mapping\try_2\try_2.ino:25:0:

C:\Users\Utsav\Documents\Arduino\libraries\arduino_583506/PS3BT.h: In function 'void loop()':

C:\Users\Utsav\Documents\Arduino\libraries\arduino_583506/PS3BT.h:196:14: error: 'virtual void PS3BT::Reset()' is protected

         void Reset();

              ^

try_2:160: error: within this context

         PS3.Reset();

                   ^

try_2:161: error: expected '}' before 'else'

         else

         ^

C:\Users\Utsav\Desktop\New manual\11 - jan - 2018\code with old mapping\try_2\try_2.ino: At global scope:

try_2:184: error: expected declaration before '}' token

 }

 ^

Multiple libraries were found for "PS3BT.h"
 Used: C:\Users\Utsav\Documents\Arduino\libraries\arduino_583506
 Not used: C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master
 Not used: C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master
 Not used: C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master
 Not used: C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0-master
exit status 1
within this context

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

i am displaying my main code shown below : - 

#define Joy_left_x PS3.getAnalogHat(LeftHatX)
#define Joy_left_y PS3.getAnalogHat(LeftHatY)
#define Joy_right_x PS3.getAnalogHat(RightHatX)
#define Joy_right_y PS3.getAnalogHat(RightHatY)
#define key_l1 PS3.getButtonPress(L1)
#define key_l2 PS3.getAnalogButton(L2)
#define key_l3 PS3.getButtonPress(L3)
#define key_r1 PS3.getButtonPress(R1)
#define key_r2 PS3.getAnalogButton(R2)
#define key_r3 PS3.getButtonPress(R3)
#define key_ps PS3.getButtonPress(PS)

#define key_up PS3.getButtonPress(UP)
#define key_down PS3.getButtonPress(DOWN)
#define key_left PS3.getButtonPress(LEFT)
#define key_right PS3.getButtonPress(RIGHT)
#define key_triangle PS3.getButtonPress(TRIANGLE)
#define key_circle PS3.getButtonPress(CIRCLE)
#define key_x PS3.getButtonPress(CROSS)
#define key_square PS3.getButtonPress(SQUARE)
#define key_select PS3.getButtonPress(SELECT)
#define key_start PS3.getButtonPress(START)
#define sen1 analogRead(A11) //A11

#include <PS3BT.h>
#include <usbhub.h>
#ifdef dobogusinclude
//#include <spi4teensy3.h>
#include <SPI.h>
#endif

USB Usb;

BTD Btd(&Usb); // You have to create the Bluetooth Dongle instance like so

PS3BT PS3(&Btd); // This will just create the instance

void setup() {
#if !defined(__MIPSEL__)
  while (!Serial);
#endif
  if (Usb.Init() == -1) {
    while (1); 
  }
}

/*************************************/
void loop() {
  Usb.Task();
    if (PS3.PS3Connected)
    {
      // main code
     /*delayMicroseconds(10);
     Reset();
     delayMicroseconds(10);
     set();*/      
    }
}

///////////////////////////////
///////////////////////////////

void Reset() 
 {
        PS3.PS3Connected = false;
        PS3.PS3MoveConnected = false;
        PS3.PS3NavigationConnected = false;
 }

 void set()
{
   PS3.PS3Connected = true;
   PS3.PS3MoveConnected = true;
   PS3.PS3NavigationConnected = true;
}

Lauszus, 2018-01-10:
@Utsav13 you should __NOT__ use the ```PS3.Reset()``` function, as it is only meant to be used internally by the library.

As I wrote in my first comment; it is better if you can figure out why your code gets stuck in the first place.

Anyway from the code it looks like you are using delays. The library requires the ```Usb.Task()``` function to be called at regular an interval. For that reason you can not use delays larger than ~1 ms. Instead use the ```millis()``` and ```micros()``` function for timekeeping. Please see the following page: https://www.arduino.cc/en/Tutorial/BlinkWithoutDelay.

Also please try to format your comments so it is more readable. Please checkout this basic guide to the Markdown syntax: https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet.

Lauszus, 2018-01-10:
@Utsav13 note that I just edited your comment above, so it is more readable.

Utsav13, 2018-01-11:
I have already done these operation using micros() .................

i listing what am i try still now 
    1.   Usb.Task()   function to be called  without any interval .
    2.  Usb.task()   function to be called  using regular interval  with millis() and micros() functions .
    3.  Usb.task()    function to be called  using regular interval  with delay() and delayMicroseconds() functions.

Now , i am describing my problem in detail....
        when my  code  get stucks ,it not takes any command from PS3 controller and it stays in last command which PS3 gives before stucks and also  at that time  my ps3 controller not disconnected .

Lauszus, 2018-01-14:
@Utsav13 please provide a minimal example where you experience the issue.