felis/USB_Host_Shield_2.0 #344: Not possible to enter AT commands in Terminal window

daniellabriotto, 2018-01-19:
Hello,

I'm trying to connect a Nokia 105 phone to an Arduino based on the project "Interfacing Arduino to a Cellular Phone" (https://www.circuitsathome.com/mcu/programming/interfacing-arduino-to-a-cellular-phone/) and using the act_terminal.ino code provided.

From the terminal window in a MacBook, I can connect to the device and get ACM ready, but I'm not able to type any AT command. Could you please advise on what might be going wrong?

Thank you!

xxxajk, 2018-01-20:
> com/mcu/programming/interfacing-arduino-to-a-cellular-phone/) and using

A couple of things could be wrong:
The phone might not be capable of using the baud rate that you specify.
The DTR DSR CTS and RTS may not be used correctly.
Stop bits , parity, data bits could be incorrect.
Might need to enable something on the phone?

daniellabriotto, 2018-01-20:
Hi xxxajk, I'm trying different baud rates changing both the program and the terminal connection. But I'm not sure where to change the other parameters you mentioned, as I'm using the example code from the USB Host Shield 2.0 library. Could you please tell me how to check them?

Lauszus, 2018-01-20:
@daniellabriotto please see these lines: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/acm/acm_terminal/acm_terminal.ino#L29-L34.

daniellabriotto, 2018-01-25:
Hi, @Lauszus! Thank you for pointing out the lines in the code. I've tried a bunch of different combination and so far we just figure out that for the connection Nokia 105, USB Host Shield, Arduino Uno and Terminal in a MacBook I have to use: 9600 when opening the connection using terminal (screen /dev/cu.usbmodem1421 9600) and Serial.begin( 9600 ). 
lc.dwDTERate didn't seem to make much of difference so I left it equal to 115200.
For lc.bDataBits, I tried using 8, 16, 32, 64 and 128 and nothing changed, I still see the Start message but can't type the AT commands.
For lc.bCharFormat and lc.bParityType I'm not sure what the possible values are.
Could anyone please advise on what to try?

Thank you!

Lauszus, 2018-02-26:
The LINE_CODING struct is documented in the source code: https://github.com/felis/USB_Host_Shield_2.0/blob/f948ffccbafdf23f34b87cf53ae14125f7d5354a/cdcacm.h#L111-L114

This gist came up after a short Google search: https://gist.github.com/plugnburn/5b2582be521944f739e1. I would recommend trying that on your computer to get it up and running there first.

From this line: https://gist.github.com/plugnburn/5b2582be521944f739e1#file-nokiatool-sh-L15 you can see that he uses two stop bits, enables odd-parity and uses 8 data bits i.e. you should set the line coding to the following:
```C++
lc.dwDTERate = 9600;
lc.bCharFormat = 2;
lc.bParityType	= 1;
lc.bDataBits = 8;
```
You might have to try other baudrates if 9600 is not working.

Lauszus, 2018-03-15:
Will close this issue, as I believe the comment above solves it. Feel free to reopen the issue again if needed.