felis/USB_Host_Shield_2.0 #345: Issue with Uno*Pro from Hobbytronics.com !

Nurrl, 2018-01-20:
Hi,
I bought an extension for Arduino UNO, which is called [Arduino UNO*Pro](http://www.hobbytronics.co.uk/arduino-uno-pro), and basically replaces the original *ATmega328p* by an *ATmega1284p*.
I tried this library with the same hardware (USB Host Shield & Arduino UNO Official Board) but with the two controllers, and the original chip works fine but the new make `Usb.Init()` returning `-1`.

Basically the pinout is the same as the Arduino UNO, (Including SPI and ports 9 and 10) so that's for sure a software issue.

Could you help me to solve this issue ?

Thanks in advance !

Lauszus, 2018-01-20:
I had a quick look at there hardware add-on and the pinout does not follow the Sanguino: https://github.com/felis/USB_Host_Shield_2.0/blob/master/avrpins.h#L723-L761.

Anyway I believe it would work if you replaced: https://github.com/felis/USB_Host_Shield_2.0/blob/master/UsbCore.h#L47 with:
```C++
typedef MAX3421e<Pb4, Pd6> MAX3421E;
```

Nurrl, 2018-01-20:
Big thanks @Lauszus, it worked ! Can you say why this works ? (Do UHS2.0 bypasses pins_arduino.h ?)

xxxajk, 2018-01-20:
Nurrl: could you test UHS 3.0 as well? Note that the interrupt pin will
have to be adjusted.

Lauszus, 2018-01-20:
@Nurrl yes it does. This is because the digitalWrite function in Arduino is quite slow, so the library writes directly to the registers.

Nurrl, 2018-01-20:
@Lauszus Awesome !

Nurrl, 2018-01-20:
@xxxajk: I didn't managed to use UHS3.0, I'll try harder later.
@Lauszus: I'm preparing a pull-request to add support for UNO*Pro in `settings.h` for manual enable.

Lauszus, 2018-01-20:
@Nurrl okay, thank you!

Nurrl, 2018-01-20:
@Lauszus: Update -> I found that `ARDUINO_AVR_UNO_PRO` was defined when building for UNO*Pro, so where should I put the code for this ?

Lauszus, 2018-01-20:
No it's better to define the pins in: https://github.com/felis/USB_Host_Shield_2.0/blob/master/avrpins.h.

Nurrl, 2018-01-20:
Oh okay, I'll do that ! @Lauszus

Lauszus, 2018-01-20:
Withed with #346.

closed 2018-01-20 by Lauszus