felis/USB_Host_Shield_2.0 #348: Serious Lag issue when combining two SPI devices.

Nurrl, 2018-01-25:
Hi, I have an issue with the USB Host Shield Library when using the UHS and an LCD screen with Hardware SPI.
The inputs triggers ~1s later and the screen has artefacts, I double checked that I use two different Slave Select pins for each device.
Could this issue come from a bad management of the Slave Select from UHSLib or just an ATMega1284 hardware limitations ?

Thanks for your help !

Lauszus, 2018-01-25:
There should not be any conflicts with the USB Host library, as it is using the Arduino SPI transactions API.

Can you share a minimal example that shows the issue?

Nurrl, 2018-01-25:
Hi @Lauszus, examples like screenshots ?

Lauszus, 2018-01-25:
No, please post the code the code into this issue.

Nurrl, 2018-01-25:
@Lauszus: Updated.

Nurrl, 2018-01-27:
Did you found something ?

Lauszus, 2018-01-28:
@Nurrl can you check how long the loop takes on average? It looks to me like you are clearing and sending the buffer in every loop. Try only doing that when the buffer actually changes.

Nurrl, 2018-01-30:
@Lauszus: Okay, I tried what you proposed, the lag disapeared, nevertherless artefacts are still on screen, particularely on line 0 and 31(or 30), where the pixels are flashing all over those rows...

Lauszus, 2018-01-30:
Please post your updated code. The ATmega1284p might simply not be fast enough.

Nurrl, 2018-01-31:
Hi @Lauszus, here is my code.

```c
if (PS3.getButtonClick(TRIANGLE)) {
      u8g2.clearBuffer();
      u8g2.setFontDirection(0);
      u8g2.setFont(u8g2_font_inb24_mf);
    
      u8g2.drawStr(0, 30, "Traingle");
      u8g2.sendBuffer();
    }
    if (PS3.getButtonClick(CIRCLE)) {
      u8g2.clearBuffer();
      u8g2.setFontDirection(0);
      u8g2.setFont(u8g2_font_inb24_mf);
    
      u8g2.drawStr(0, 30, "Circle");
      u8g2.sendBuffer();
    }
    if (PS3.getButtonClick(CROSS)) {
      u8g2.clearBuffer();
      u8g2.setFontDirection(0);
      u8g2.setFont(u8g2_font_inb24_mf);
    
      u8g2.drawStr(0, 30, "Cross");
      u8g2.sendBuffer();
    }
    if (PS3.getButtonClick(SQUARE)){
      u8g2.clearBuffer();
      u8g2.setFontDirection(0);
      u8g2.setFont(u8g2_font_inb24_mf);
    
      u8g2.drawStr(0, 30, "Square");
      u8g2.sendBuffer();
    }

    if(PS3.getButtonClick(PS))
      PS3.disconnect();
```

Lauszus, 2018-02-26:
@Nurrl is this code still really slow? Can you try to connect the PS3 controller via USB instead? Also do you actually have to clear the buffer, set the font direction and the font every time. Since you are using the same every time I don't think it would be necessary? You could just add spaces in the end of a string in order to erase any previous written character. If the speed is still an issue, then I would recommend using a Teensy 3.6 or similar.

Nurrl, 2018-02-27:
I think I'll refactor this code to update the screen only when needed, also, is there a speed difference between the full-buffer mode and the simple-buffer mode ?

Lauszus, 2018-03-01:
@Nurrl I assume that you are referring to the U8G2 library? I'm not familiar with that, so it is better to ask the author of that library.

Lauszus, 2018-03-15:
I will close this issue, as I believe it has been solved.

Lauszus, 2018-03-16:
Actually #354 might fix this, so I am reopening the issue.

@Nurrl could you please replace the following line: https://github.com/felis/USB_Host_Shield_2.0/blob/03933edb41bd277824181dc79454175dfc714dd4/usbhost.h#L464 with:
```C++
regWr(rHIEN, bmCONDETIE); //connection detection
```
And then see if it makes any realworld difference?

Nurrl, 2018-03-17:
Of course, I'll try as soon as possible ! @Lauszus

Nurrl, 2018-03-24:
Hi @Lauszus, I tested the fix, refactoring the code made the lag disapear, but even with the fix you proposed, some artefacts are still there.

Nurrl, 2018-03-24:
[image]

Lauszus, 2018-03-28:
Thanks for testing @Nurrl. I believe you are simply just seeing the limitation of the ATMega1284. I would recommend using a Teensy 3.6, as it should be pretty much a drop-in replacement.

Nurrl, 2018-03-28:
I'll stay with my ATMega1284p for now since this is a school project, I'll include this issue in my report because we need to tell what were the issues we encontered, etc.

I close the issue because it seems unresolvable by software update. If I find any fix, I'll reopen this issue to discuss about it.

closed 2018-03-28 by Nurrl