felis/USB_Host_Shield_2.0 #349: Multiple Slave Devices on Arduino Mega 2560 with USB Host Shield

xmdfb, 2018-02-03:
Hi, 

I am currently experiencing a weird issue where with the USB Host Shield attached to the Arduino Mega 2560, I am not able to communicate with another SPI Device (its not a shield so I have wired it using pins 50 to 53). 

So the chip select pins are definitely different, but still unable to communicate. The wiring is correct as when I disconnect the host shield, I can perfectly communicate with the SPI Device. 

Any ideas why the USB Host shield is blocking the SPI signal?

xxxajk, 2018-02-03:
Make sure that whatever you are using on SPI use begin/end transaction as
shown here

https://github.com/felis/USB_Host_Shield_2.0/blob/03933edb41bd277824181dc79454175dfc714dd4/usbhost.h#L173

<https://github.com/felis/USB_Host_Shield_2.0/issues/349>, or mute the
thread

xmdfb, 2018-02-03:
Thanks for your reply. The code that I am testing with does not involve the USB Host shield, it literally just controls the SPI LCD display. However, I do have the USB Host shield attached to the Mega 2560. Do I still need to include begin/end transaction?

xxxajk, 2018-02-03:
No, but you should set SS used on the USB Host Shield HIGH to ensure it is
disabled.

xmdfb, 2018-02-03:
Yeah, I tried that actually. I put pin 10 to HIGH, but it still didn't work. Is there anything else I need to do to disable the SPI on the USB Host shield? (The LCD display works on SPI as soon as I take off the host shield).

xxxajk, 2018-02-03:
Check with an oscilloscope if you have one, you may just need to add
pull-up resistors on the SPI lines.... Or pulldowns... What value to use
exactly will take some experimenting.

Lauszus, 2018-02-04:
@xmdfb can you send me the link to the LCD and library you are using? My guess would be that the LCD library does not use the transaction API. Another issue could be if they use the same SS pin. What SS pin is the LCD using?

Btw there is no need to manually set the SS pin for the USB Host shield, as this is handled by the library.

Lauszus, 2018-03-15:
Closing this issue due to lack of feedback. Please provide more details and reopen the issue if needed.

closed 2018-03-15 by Lauszus