felis/USB_Host_Shield_2.0 #350: Usb Host acm

peppegti, 2018-02-08:
Hi, my name is Peppe.
I want use my board mega + usb host for read a gps usb.
The code is the acm_terminal and work well, but I need save the string read from the gps.
now the code, read buf and print the buf[i], but if I want save the buf in the string strgps not have all 
characters because the buf is initialized uint8_t buf[64] and the string nmea is about 75.
I try to increase bu[75] but not work and after 10 or 5 or 2 (random) it freezes and does not exceed `if( rcvd ) { //more than zero bytes received`

first try:
`String strNMEA(reinterpret_cast<char *> (buf));`
in this case output is:
$GNGGA,090458.00,0000.56665,N,0000.63494,E,1,09,1.18,964.4,M,37

second try:
`if( rcvd ) { //more than zero bytes received
              for(uint16_t i=0; i < rcvd; i++ ) {
                //Serial.print((char)buf[i]); //printing on the screen
                strNMEA+=((char)buf[i]);
                
              }
            }
            Serial.println(strNMEA);
            strNMEA="";`

in this case output is:
$GNGGA,093410.00,0000.57042,N,0000.63206,E,1,08,1.22,969.2,M,37
.5,M,,*42

post all code:
`#include <cdcacm.h>
#include <usbhub.h>

#include "pgmstrings.h"

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

class ACMAsyncOper : public CDCAsyncOper
{
public:
    uint8_t OnInit(ACM *pacm);
};

uint8_t ACMAsyncOper::OnInit(ACM *pacm)
{
    uint8_t rcode;
    // Set DTR = 1 RTS=1
    rcode = pacm->SetControlLineState(3);

    if (rcode)
    {
        ErrorMessage<uint8_t>(PSTR("SetControlLineState"), rcode);
        return rcode;
    }

    LINE_CODING  lc;
    lc.dwDTERate  = 9600;
    lc.bCharFormat  = 0;
    lc.bParityType  = 0;
    lc.bDataBits  = 8;

    rcode = pacm->SetLineCoding(&lc);

    if (rcode)
        ErrorMessage<uint8_t>(PSTR("SetLineCoding"), rcode);

    return rcode;
}

USB     Usb;
//USBHub     Hub(&Usb);
ACMAsyncOper  AsyncOper;
ACM           Acm(&Usb, &AsyncOper);

void setup()
{
  Serial.begin( 9600 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
      Serial.println("OSCOKIRQ failed to assert");

  delay( 200 );
}

void loop()
{
    Usb.Task();

    if( Acm.isReady()) {
       uint8_t rcode;

       
       delay(50);
String strNMEA;
        /* reading the phone */
        /* buffer size must be greater or equal to max.packet size */
        /* it it set to 64 (largest possible max.packet size) here, can be tuned down
        for particular endpoint */
        uint8_t  buf[64];
        uint16_t rcvd = 64;
        memset (buf, 0x00, sizeof (buf));
        rcode = Acm.RcvData(&rcvd, buf);
         if (rcode && rcode != hrNAK)
            ErrorMessage<uint8_t>(PSTR("Ret"), rcode);

            if( rcvd ) { //more than zero bytes received
              for(uint16_t i=0; i < rcvd; i++ ) {
                //Serial.print((char)buf[i]); //printing on the screen
                strNMEA+=((char)buf[i]);//second try
                
              }
            }
            Serial.println(strNMEA);
            strNMEA="";
            //First try
            //String strNMEA(reinterpret_cast<char *> (buf));
            // Serial.println(strNMEA);
            //strNMEA="";
        delay(10);
    }//if( Usb.getUsbTaskState() == USB_STATE_RUNNING..
    
}
`

Lauszus, 2018-02-24:
Just remove the line:
```C++
while (!Serial);
```
From the example.

peppegti, 2018-02-24:
Hi 
I have try but not work

Lauszus, 2018-02-24:
You need to provide me with some more information then. Can you provide a minimal example that shows the behaviour?

peppegti, 2018-02-24:
It s possible need to change the power option?
I have an arduino mega with sd shield and usb host.
In the usb I connect the gps with the code acm.

Lauszus, 2018-03-06:
Please provide a minimal example that shows the behaviour.

peppegti, 2018-03-06:
The external power is the first step ok, bit the shield have need more reboot for start. I Think the problem is the initialition

Lauszus, 2018-03-07:
As I have said two times already you need to post your source code if you want me to help you out.

peppegti, 2018-03-15:
`#include <cdcacm.h>
#include <usbhub.h>

#include "pgmstrings.h"

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

class ACMAsyncOper : public CDCAsyncOper
{
public:
    uint8_t OnInit(ACM *pacm);
};

uint8_t ACMAsyncOper::OnInit(ACM *pacm)
{
    uint8_t rcode;
    // Set DTR = 1 RTS=1
    rcode = pacm->SetControlLineState(3);

    if (rcode)
    {
        ErrorMessage<uint8_t>(PSTR("SetControlLineState"), rcode);
        return rcode;
    }

    LINE_CODING	lc;
    lc.dwDTERate	= 9600;
    lc.bCharFormat	= 0;
    lc.bParityType	= 0;
    lc.bDataBits	= 8;

    rcode = pacm->SetLineCoding(&lc);

    if (rcode)
        ErrorMessage<uint8_t>(PSTR("SetLineCoding"), rcode);

    return rcode;
}

USB     Usb;
//USBHub     Hub(&Usb);
ACMAsyncOper  AsyncOper;
ACM           Acm(&Usb, &AsyncOper);

void setup()
{
  Serial.begin( 9600 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
      Serial.println("OSCOKIRQ failed to assert");

  delay( 200 );
}

void loop()
{
    Usb.Task();

    if( Acm.isReady()) {
       uint8_t rcode;

       /* reading the keyboard */
       if(Serial.available()) {
         uint8_t data= Serial.read();
         /* sending to the phone */
         rcode = Acm.SndData(1, &data);
         if (rcode)
            ErrorMessage<uint8_t>(PSTR("SndData"), rcode);
       }//if(Serial.available()...

       delay(50);

        /* reading the phone */
        /* buffer size must be greater or equal to max.packet size */
        /* it it set to 64 (largest possible max.packet size) here, can be tuned down
        for particular endpoint */
        uint8_t  buf[64];
        uint16_t rcvd = 64;
        rcode = Acm.RcvData(&rcvd, buf);
         if (rcode && rcode != hrNAK)
            ErrorMessage<uint8_t>(PSTR("Ret"), rcode);

            if( rcvd ) { //more than zero bytes received
              for(uint16_t i=0; i < rcvd; i++ ) {
                Serial.print((char)buf[i]); //printing on the screen
              }
            }
        delay(10);
    }//if( Usb.getUsbTaskState() == USB_STATE_RUNNING..
}

`

peppegti, 2018-03-15:
[image]

Now I try with a mini board, but same problem

peppegti, 2018-03-15:
[image]

Not read acm terminal

sometimes 
OSCOKIRQ failed to assert

Lauszus, 2018-03-15:
As I wrote before you need to remove:

```C++
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
```
If you want to use the code without attaching the serial terminal.

What Arduino are you using? Is it running at 3.3V or 5V?

peppegti, 2018-03-15:
I use the arduino nano with 5v

Lauszus, 2018-03-15:
You can not use that directly with the USB Host Mini shield, as it is 3.3V only!

You might have damaged the MAX3421E chip.

Please read: https://www.circuitsathome.com/usb-host-shield-hardware-manual.

peppegti, 2018-03-15:
you can see in the previus photo the yellow circle. I have interrupt the circuit 3.3

Lauszus, 2018-03-15:
That does not matter! The high level of the GPIO on the Arduino will still be 5V and the MAX3421E is NOT 5V tolerant.

I would recommend using the full size version of the shield, as the mini version is meant for advanced users.

peppegti, 2018-03-15:
I have the normal shield with the arduino mega but or work the usb or  the sd shield. 
when I connect all the code block
[image]

peppegti, 2018-03-15:
you think the problem for the mini is the voltage pin?

Lauszus, 2018-03-15:
So is the Arduino Mega working without the SD card?

No you can NOT use the Arduino nano directly with the Usb Host mini shield, as it does not have logic level converters!

peppegti, 2018-03-15:
Ok but if add the logical level convert it can work?
Yes, after I connect the sd shield the code block.

Lauszus, 2018-03-15:
What pins have you connected the SD card to and are you using the official SD card library or something else?

peppegti, 2018-03-15:
the problem is the pin 50 (miso)

Lauszus, 2018-03-15:
Okay, what is the problem? Please describe it, as other people might find it useful in the future.

peppegti, 2018-03-15:
in my project I have to build a board that reads a usb gps and saves on sd.
the library is the classic spi.h and sd.h

Lauszus, 2018-03-15:
Okay. I will close this issue now, as I believe this issue has been solved.

closed 2018-03-15 by Lauszus

peppegti, 2018-03-15:
Why you close? the problem it s not solve.
[https://www.pjrc.com/teensy/td_libs_USBHostShield.html](url) in this case use a board with 5 volts

Lauszus, 2018-03-15:
As the page you just linked to also says in the first couple of lines is that the "Teensy MUST be modified for 3.3 volts" and then links to the following site: https://www.pjrc.com/teensy/3volt.html.

I would recommend getting a Teensy 3.6, as it is already running at 3.3V and is more powerful.

Sorry, but Github issues are not meant for helping you out with your personal projects, but only if there is problems with the library or for feature requests. For general help with Arduino I would recommend visiting the official forum: <http://forum.arduino.cc>.

peppegti, 2018-03-15:
ok ok , but why the full size not work? in the serial term have only the same error: OSCOKIRQ failed to assert  
I read the problem its a current? for this reasons the solution is a capacitor. it s really?

Lauszus, 2018-03-15:
Sorry, but you have no described your project/code in any detail, so there is no way for me to help you out. Start by getting the shield working without the SD card, then get the SD card working without the USB host shield and finally try to combine two. It's important that you do one step at a time.