felis/USB_Host_Shield_2.0 #351: PS4 pairing implemented in code

puttley, 2018-02-09:
Hello,

I'm having no issues pairing the PS4 with BT Dongle. However, is it possible to pair the PS4 with a dongle if I were to add a "pairing button" to my UNO instead of having to first pair, then comment / un-comment lines in the example sketch? I have tried calling PS4.pair() within code but only get "HCI Command Failed: 0C" in serial monitor debug. 

Many thanks..

Lauszus, 2018-02-24:
Please enable serial debugging: https://github.com/felis/USB_Host_Shield_2.0#enable-debugging and post the full output.

puttley, 2018-02-24:
Issue is resolved - Paring is successful if I always call disconnect before PS4.pair().

closed 2018-02-24 by puttley

Lauszus, 2018-02-26:
Good to hear you got it working :)

microbmen, 2023-08-05:
Thank you so much for this post.  I'm able to attempt connect without pair, unless a toggle switch is enabled which then does a disconnect/pair.  Works perfect.  

Code for anyone else trying to figure this out.  Its huge to be able to pair without modifying and rewriting code to chips.

`
// try to connect to previously paired devices
PS4BT PS4(&Btd);

void setup() {
  // set the slaveSelectPin as an output:
  pinMode(slaveSelectPin, OUTPUT);

  // initialize SPI:
  SPI.begin();

  Serial.begin(115200);
#if !defined(__MIPSEL__)
  while (!Serial)
    ;  // Wait for serial port to connect - used on Leonardo, Teensy and other boards with buiL2-in USB CDC serial connection
#endif
  if (Usb.Init() == -1) {
    Serial.print(F("\r\nOSC did not start"));
    while (1)
      ;  //halt
  }
// read the voltage from the 'pairing button'
  float PairPinVal = float(analogRead(PairPin));

  Serial.println(F("\r\nRobot Controller Bluetooth Library Started"));

//if button is pushed there will be some value 0-1023
  if (PairPinVal >= 100) {
    Serial.println("Pairing...");
    PS4.disconnect();
    PS4.pair();
  }

}`