felis/USB_Host_Shield_2.0 #353: How do we rename SS and Int pins to A5 and A0?

jsweet13, 2018-02-20:
Hello,

I'm browsing through all files in the 2.0 library to find the SS and Int pinout for the Arduino Uno but not really having any luck.

I broke the connections and rerouted SS to pin A5 and rerouted INT to A0 on my Arduino Uno so I can display chars on a tftlcd display.

How do I reflect A5 and A0 pins in the USBcore.h file?

I see and changed this line to:
typedef MAX3241e<PA0, PA5> MAX3241E;

Is this correct? And are there regular variable names I can use to change the PIN numbers dynamically in code?

Thanks for your help.

Lauszus, 2018-02-20:
Please read: https://github.com/felis/USB_Host_Shield_2.0#interface-modifications.

The analog pins A0-A5 are referred to as P14-19, so to use A5 and A0 you should use the following:

```C++
typedef MAX3421e<P14, P19> MAX3421E;
```

jsweet13, 2018-02-25:
Sounds great. I also found the analog pins were declared under a pinout section for the Intel Galileo in one of the pin definition header files and made changes accordingly. It's working pretty good.

Is there a way to bypass the USB.Task continuous loop as shown in the bootkeyboard example and scanf() with stdio.h to get individual chars?

I tried breaking up the class to just list the functions but when I do that, oemtoascii() function becomes invalid and the redefined for USB and prs become invalid.

I'd like to find a way to catch chars with scanf() using the USB keyboard as stdio input without the USB.Task function.

I also noticed USB is an Arduino IDE keyword but think that refers to the Uart to USB bridge.

I was trying to print individual chars to an LCD display with a custom font I made but when I send the char to my function printchar() in the keydown() that states oemtoascii((char)c), my code keeps rebooting to setup(), even if I move the USB.Task() to another function or send the program to loop() or do a while(1){};

How do I stop usb.task from being continuous and scanf() for the chars?

Thanks

Lauszus, 2018-02-26:
The function ```OnKeyDown``` will be called every time a key is pressed. I don't understand why you want to use scanf to read the keyboard. You can just implement similar functionality yourselves.

Please share your code, so I can have a look at it.

Sorry, but it sounds like you have not really understood how this library works. You will have to call ```Usb.Task()```, as this is how the library reads from the different USB devices.

jsweet13, 2018-02-26:
I was trying to figure out how to use the functions outside of the class but it does work as is.

Will try to post some code in a bit.

I might try a function call embedding the OEMtoAscii(mod, key) into my custom function PrintChar that prints pixels to a display to form:

PrintChar(OEMtoAscii(mod, key));

Void PrintChar(uint8_t c){
//pointer to char[] artwork
};

I'm guessing mod and key variables are both needed to convert the keystrokes to ASCII values?

The scanf() seems to work with chars coming from the serial terminal, but I was wanting to switch() case each char after getting the chars to print with my font.

I might have to make all the variables public so my functions can see them.