felis/USB_Host_Shield_2.0 #354: Interrupt handling

TomsCircuits, 2018-02-21:
I am currently working my way through the source code of this library, because I am trying to understand how to use te MAX3421e. Big thanks to all the authors!

I noticed something odd. In both versions of MAX3421e::Init() the SOF interrupt is enabled (bmFRAMEIE = 1). This means that the INT pin gets activated.

Then, in MAX3421e::Task this is detected and the IntHandler is invoked. But there, all frame INT handling is commented out, so that the FRAMEIRQ bit is never reset. Instead, the CONDETIRQ bit is tested. 

In summary this means that the Task handler invokes the IntHandler every time without any need.

Not sure if I am right. I am not a real programmer (tm) :-)

xxxajk, 2018-02-22:
Doesn't matter, as UHS2 is polled anyway?

TomsCircuits, 2018-02-22:
Thanks for the reply. Well yes, UHS2 is polled. But in a way, this is my point. In USB::Task the MAX3421e's _INT pin_ is polled. And since UHS2 only cares for CONDETIRQs, the INT pin should not be activated on a FRAMEIRQ.

In MAX3421e::Init, the Interrupt Enable register HIE is set like this:
        regWr(rHIEN, bmCONDETIE | bmFRAMEIE); //connection detection

Why is bmFRAMEIE bit set, when it is never processed? It leads to an unnecessary invocation of the interrupt handler each time the USB::Task is called. In the INT handler the HIRQ register is read but not reset. A lot of activity that makes no sense, but costs CPU time.

Because the FRAMEIRQ bit is never reset, effectively during each USB::Task the HIRQ register is polled, after the INT pin is polled.

TomsCircuits, 2018-02-22:
Oh, and by the way, the INT handler writes 0 to HIRQ with this line:

`regWr(rHIRQ, HIRQ_sendback);

because HIQ_sendback remains 0 if FRAMEIRQ triggered the INT handler. Writing 0 to this register is the equivalent to resetting nothing, so this is also unnecessary.

xxxajk, 2018-02-22:
I'll try to take a look at this over the weekend, provided that I get the
free time to do so.
I think you might be on to something...

Lauszus, 2018-03-15:
@TomsCircuits any updates?

TomsCircuits, 2018-03-15:
Well, for a test I modified the code so that bmFRAMEIE would not be set. This seemingly made no difference as far as the function of the library is concerned. I did however verify with a logic analyser that there was a massive reduction of activity on the SPI bus.

But i had hoped that someone would be able to review this issue independently from me.

Lauszus, 2018-03-16:
@TomsCircuits this might actually solve this issue #348. I have asked the user to try your fix and see if it makes any difference in his application.