felis/USB_Host_Shield_2.0 #355: How to know if PS3 controller disconnects

hanuxe, 2018-02-27:
I'm using PS3 controllers to teleoperate small robots, but they behave erratically when the controller disconnects. That is, when the controller times out due to lack of use, gets too far from the Bluetooth dongle, or runs out of battery, there doesn't seem to be a way to know that the controller has disconnected and to automatically stop the robot. 

My code currently checks `PS3.PS3Connected` to see if the controller is connected, but `PS3.PS3Connected = 1` even when the controller has disconnected and shut off. It does succeed at detecting whether the controller has connected initially, before the disconnects, at least.

Is there another way to detect a disconnect?

Thanks!

xxxajk, 2018-02-27:
USB or bluetooth? I don't have this controller, but I can check the code and make sure it is detecting the case properly.

hanuxe, 2018-02-28:
Bluetooth. 
For reference, my controller setup uses the following hardware:
* PS3 controller (https://www.walmart.com/ip/Bluetooth-Wireless-Black-Game-Controller-for-PlayStation-3-PS3-USB-Wired-Available/50175929)
* Bluetooth dongle, TRENDnet (https://www.amazon.com/TRENDnet-Bluetooth-100Meters-Compatible-TBW-106UB/dp/B002AQSTXM?th=1)
* USB host shield, TKJ (http://shop.tkjelectronics.dk/product_info.php?products_id=43)
* Arduino Mega (https://www.amazon.com/keyes-development-board-compatible-arduino/dp/B01FVGY1KM/)

(I'll post my code momentarily.)

Lauszus, 2018-03-01:
@hanuxe no the current library does not detect if the controller is suddenly disconnected. This has also been brought up in the past.

What we should do is monitor if the controller stops sending data or not and then set PS3Connected to false.

Lauszus, 2018-03-15:
I will close this issue, as I believe it has been solved.

closed 2018-03-15 by Lauszus