felis/USB_Host_Shield_2.0 #356: Possibly Defective Board?

SkyWhe, 2018-02-28:
Hello I picked up a host board 2.0 and I am having a hell of a time getting it to connect to a PS4 controller.

I am also using 2 different BT dongles that have been confirmed to work by several posters. **(TRENDNET TBW-106UB and 107UB)**

I have all the libraries installed and working, I can compile and upload the code. However, very rarely does it even turn the dongle on. A blue light can be seen in my third BT dongle at all times when plugged into any usb port, even non-data cables causes it to light up so this is not only something that shows during transmission.

**To remedy this I have attempted:**
-  To power the arduino I am using from an external power source but it caused no change in functionality. 
- I have tried multiple different Arduino unos and cables.

Another issue I have is I **randomly get the error message "OSC did not start".** I see from the code that this is when the board itself fails to initialize. Sometimes power cycling the board fixes this, but it seems to be totally at random. 

**To investigate this I tried to turn on the debugging statements by changing the ENABLE_UHS_DEBUGGING to 1 in the settings.h file**. Strangely enough, the very next time I tried to upload my code and compile I got the debug information. But then never again. No matter how many times I changed the value of the ENABLE_UHS_DEBUGGING. I made sure I was editing the right file by deleting a chunk of the settings.h and making sure my code would fail to compile. **I also tried changing my baud rate to make sure I wasn't having an issue there.**

All of these issues aside, in the event I don't get the "OSC" error, and the dongle decides to turn on, when I follow the instructions to pair my PS4 controller (putting PS4 controller in pair mode, starting code running on arduino) it just does nothing. The PS4 controller blinks until it times out and I am just staring at a serial console saying "PS4 Bluetooth Library Started".

At this point I am at a total loss and it can't find anyone that has had any of the issues I am having. Is there something I am just totally missing? **I should mention that I have put all 3 of the dongles I am using into my computer and successfully paired my PS4 controller with them.**

xxxajk, 2018-02-28:
"OSC did not start" ... this is all from my experience.
1: 3.3v supply is noisy, basically the 3.3v regulator is bad
2: bad contact, bad solder joint on the usb host shield
3: defective board.

Lauszus, 2018-03-01:
@SkyWhe please send a link to the shield you are using. Also please run the following example: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/board_qc/board_qc.ino and post the output.

SkyWhe, 2018-03-01:
> examples/board_qc/board_qc.ino and post the output.

https://www.sparkfun.com/products/9947

I will try the QC code.

Lauszus, 2018-03-01:
You need to connect pin 7 to the reset pin on the Arduino. You also need to connect an external power supply.

xxxajk, 2018-03-01:
Note that this shield (sparkfun) is only compatible with the uno, I
think.... The official shield works with just about everything else,
without modification. The only exception I have found are digilent boards,
which have screwed up on the SPI connector.

SkyWhe, 2018-03-01:
Thank you both for the suggestions. I would like to add I have strictly been trying to get the board to work with an UNO. I have also made sure to try operation with an external power supply.

Here is the output of the QC code with an external power source and pin 7 bridged to the reset pin.

Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K................ SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Value written: 00 Value read: FF

xxxajk, 2018-03-02:
That's okay to fail that test. Hit the send button to finish up the test.

Lauszus, 2018-03-15:
I will close this issue, as I believe it has been solved.

closed 2018-03-15 by Lauszus

SkyWhe, 2018-03-15:
Yeah sorry for not getting back. I ran the complete test and it passed!

And then , without changing anything, literally nothing, I reloaded my code
and it worked.

I truly cannot explain what the issue was, but somehow by running the QC
code it was resolved. Thank you all for your time.

Lauszus, 2018-03-15:
Glad to hear you got it working :)