felis/USB_Host_Shield_2.0 #357: XBOX ONE PC Adapter with Host Shield?

joptimus, 2018-02-28:
Hello,

I implemented the code for XBOX 360 controller and all works well. I have a XBOX ONE Controller and am trying to use a XBOX ONE Wireless PC adapter in the USB Host Shield.  Is this dongle supported?  If not can you direct me to a correct dongle that will work with the Xbox One controller?

This is the adapter I am trying to use

https://www.gamestop.com/pc/accessories/xbox-one-wireless-pc-adapter/125488

Lauszus, 2018-03-06:
No the dongle is currently not support. If you donate one to the project including a Xbox One controller, then I will add support for it.

Soultaker, 2018-04-25:
@Lauszus Currently working on this and hoping to add to the library. The code is based on the XBOXRECV class. Any tips?

Soultaker, 2018-04-25:
To note i used the ACM Terminal and USB Desc but nothing real information printed on Serial Monitor
What printed was Start or something similar. Less than 4 lines printed.

Soultaker, 2018-04-25:
XBOX USB Library StartedECV Init
BTD Init
Addr: 01
XBOX USB Library Started
XONERECV Init
BTD Init
Addr: 01
Xbox Wireless Receiver Connected

Is what I get when I run current code

Soultaker, 2018-04-27:
Running the USB DESC I get the following:

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0201
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		045E
Product ID:		02E6
Revision ID:		0100
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		004A
Num.intf:		01
Conf.value:		01
Conf.string:		02
Attr.:			A0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		08
Intf. Class:		FF
Intf. Subclass:		FF
Intf. Protocol:		FF
Intf.string:		02

Endpoint descriptor:
Endpoint address:	84
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	85
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	08
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	04
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	05
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	06
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	07
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	09
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)

Soultaker, 2018-04-30:
@Lauszus Do you mean the actual controller or the dongle for the controller?

Lauszus, 2018-04-30:
@Soultaker sorry I mixed up this issue with #378, so please stay with the XBOXRECV code.

Anyway you should modify the endpoint entry to match the descriptor above: https://github.com/felis/USB_Host_Shield_2.0/blob/f948ffccbafdf23f34b87cf53ae14125f7d5354a/XBOXRECV.cpp#L191-L241.

Soultaker, 2018-04-30:
So 82 also means an input endpoint with address 2?

Lauszus, 2018-04-30:
Yes exactly :)

Soultaker, 2018-04-30:
One other thing
For the line
epInfo[ XBOX_INPUT_PIPE_1 ].epAddr = 0x01;

Was the 0x01 something arbitrary or did it come from somewhere in the descriptor?

Lauszus, 2018-04-30:
No it's not arbitrary! That is the equivalent of ```Endpoint address: 81```.

Soultaker, 2018-04-30:
So the reason why I thought it was arbitrary was because the addresses in the xboxrecv were odd numbers

Soultaker, 2018-04-30:
@Lauszus 
Whereas the ep address I pulled from a official xbox360 is as such 

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	FF
Device Protocol:	FF
Max.packet size:	08
Vendor  ID:		045E
Product ID:		0291
Revision ID:		0107
Mfg.string index:	00
Prod.string index:	00
Serial number index:	00
Number of conf.:	01
Total length truncated to 256 bytes
Configuration descriptor:
Total length:		0141
Num.intf:		08
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		82

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		5D
Intf. Protocol:		81
Intf.string:		00
Unknown descriptor:
Length:		14
Type:		22
Contents:	000113811D001701020813010C000C0102080705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0020
Polling interval:	01

Endpoint descriptor:
Endpoint address:	01
Attr.:			03
Max.pkt size:		0020
Polling interval:	08

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		5D
Intf. Protocol:		82
Intf.string:		00
Unknown descriptor:
Length:		0C
Type:		22
Contents:	000101820040010220000705

Endpoint descriptor:
Endpoint address:	82
Attr.:			03
Max.pkt size:		0020
Polling interval:	02

Endpoint descriptor:
Endpoint address:	02
Attr.:			03
Max.pkt size:		0020
Polling interval:	04

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		5D
Intf. Protocol:		81
Intf.string:		00
Unknown descriptor:
Length:		14
Type:		22
Contents:	000113831D001701020813030C000C0102080705

Endpoint descriptor:
Endpoint address:	83
Attr.:			03
Max.pkt size:		0020
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			03
Max.pkt size:		0020
Polling interval:	08

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		5D
Intf. Protocol:		82
Intf.string:		00
Unknown descriptor:
Length:		0C
Type:		22
Contents:	000101840040010420000705

Endpoint descriptor:
Endpoint address:	84
Attr.:			03
Max.pkt size:		0020
Polling interval:	02

Endpoint descriptor:
Endpoint address:	04
Attr.:			03
Max.pkt size:		0020
Polling interval:	04

Interface descriptor:
Intf.number:		04
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		5D
Intf. Protocol:		81
Intf.string:		00
Unknown descriptor:
Length:		14
Type:		22
Contents:	000113851D001701020813050C000C0102080705

Endpoint descriptor:
Endpoint address:	85
Attr.:			03
Max.pkt size:		0020
Polling interval:	01

Endpoint descriptor:
Endpoint address:	05
Attr.:			03
Max.pkt size:		0020
Polling interval:	08

Interface descriptor:
Intf.number:		05
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		5D
Intf. Protocol:		82
Intf.string:		00
Unknown descriptor:
Length:		0C
Type:		22
Contents:	000101860040010620000705

Endpoint descriptor:
Endpoint address:	86
Attr.:			03
Max.pkt size:		0020
Polling interval:	02

Endpoint descriptor:
Endpoint address:	06
Attr.:			03
Max.pkt size:		0020
Polling interval:	04

Interface descriptor:
Intf.number:		06
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		5D
Intf. Protocol:		81
Intf.string:		00
Unknown descriptor:
Length:		14
Type:		22
Contents:	0001000108C600010000010100000012015D045E

Addr:1(0.0.1)

Soultaker, 2018-05-01:
@Lauszus Regarding endpoint descriptors, how would I which one to use?

xxxajk, 2018-05-01:
A few bits here for you...
I've taken the relevant bits from [here](http://www.usbmadesimple.co.uk/ums_3.htm)

Endpoints

Each USB device has a number of endpoints. Each endpoint is a source or sink of data. A device can have up to 16 OUT and 16 IN endpoints.

OUT always means from host to device.

IN always means from device to host.

Endpoint 0 is a special case which is a combination of endpoint 0 OUT and endpoint 0 IN, and is used for controlling the device.
Pipe

A logical data connection between the host and a particular endpoint, in which we ignore the lower level mechanisms for actually achieving the data transfers.
Transactions

Simple transfers of data called 'Transactions' are built up using packets. 

You should follow up reading from [here](http://www.usbmadesimple.co.uk/ums_4.htm)

Soultaker, 2018-05-01:
@xxxajk Thanks for the information.

xxxajk, 2018-05-01:
@Soultaker You are quite welcome.

Soultaker, 2018-05-01:
So I am still confused as to the numbers that the pipes are assigned to in the .h files of the various examples and the the addresses they are assigned in their respective .cpp files.

Soultaker, 2018-05-01:
For clarification as to where I am confused here is an example

Interface descriptor:
Intf.number: 00
Alt.: 00
Endpoints: 08
Intf. Class: FF
Intf. Subclass: FF
Intf. Protocol: FF
Intf.string: 02

Endpoint descriptor:
Endpoint address: 84
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Endpoint descriptor:
Endpoint address: 85
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Endpoint descriptor:
Endpoint address: 08
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Soultaker, 2018-05-01:
I see 3 endpoints with 2 being inputs and 1 being output.
Would I define them as such?

#define CONTROL_PIPE        0
#define OUTPUT_PIPE         1
#define INPUT_PIPE_1          2
#define INPUT_PIPE _2        3

Soultaker, 2018-05-01:
Just to make sure I am on the right track, is this what is meant by modifying the endpoints to match the descriptors? 

epInfo[ XONE_INPUT_PIPE_84 ].epAddr = 0x04; // XBOX 360 report endpoint - poll interval 1ms
        epInfo[ XONE_INPUT_PIPE_84 ].epAttribs = USB_TRANSFER_TYPE_INTERRUPT;
        epInfo[ XONE_INPUT_PIPE_84 ].bmNakPower = USB_NAK_NOWAIT; // Only poll once for interrupt endpoints
        epInfo[ XONE_INPUT_PIPE_84 ].maxPktSize = EP_MAXPKTSIZE;
        epInfo[ XONE_INPUT_PIPE_84 ].bmSndToggle = 0;
        epInfo[ XONE_INPUT_PIPE_84 ].bmRcvToggle = 0;
        
        epInfo[ XBOX_OUTPUT_PIPE_4 ].epAddr = 0x04; // XBOX 360 output endpoint - poll interval 8ms
        epInfo[ XBOX_OUTPUT_PIPE_4 ].epAttribs = USB_TRANSFER_TYPE_INTERRUPT;
        epInfo[ XBOX_OUTPUT_PIPE_4 ].bmNakPower = USB_NAK_NOWAIT; // Only poll once for interrupt endpoints
        epInfo[ XBOX_OUTPUT_PIPE_4 ].maxPktSize = EP_MAXPKTSIZE;
        epInfo[ XBOX_OUTPUT_PIPE_4 ].bmSndToggle = 0;
        epInfo[ XBOX_OUTPUT_PIPE_4 ].bmRcvToggle = 0;

        epInfo[ XONE_INPUT_PIPE_85 ].epAddr = 0x05; // XBOX 360 report endpoint - poll interval 1ms
        epInfo[ XONE_INPUT_PIPE_85 ].epAttribs = USB_TRANSFER_TYPE_INTERRUPT;
        epInfo[ XONE_INPUT_PIPE_85 ].bmNakPower = USB_NAK_NOWAIT; // Only poll once for interrupt endpoints
        epInfo[ XONE_INPUT_PIPE_85 ].maxPktSize = EP_MAXPKTSIZE;
        epInfo[ XONE_INPUT_PIPE_85 ].bmSndToggle = 0;
        epInfo[ XONE_INPUT_PIPE_85 ].bmRcvToggle = 0;
        
        epInfo[ XBOX_OUTPUT_PIPE_5 ].epAddr = 0x05; // XBOX 360 output endpoint - poll interval 8ms
        epInfo[ XBOX_OUTPUT_PIPE_5 ].epAttribs = USB_TRANSFER_TYPE_INTERRUPT;
        epInfo[ XBOX_OUTPUT_PIPE_5 ].bmNakPower = USB_NAK_NOWAIT; // Only poll once for interrupt endpoints
        epInfo[ XBOX_OUTPUT_PIPE_5 ].maxPktSize = EP_MAXPKTSIZE;
        epInfo[ XBOX_OUTPUT_PIPE_5 ].bmSndToggle = 0;
        epInfo[ XBOX_OUTPUT_PIPE_5 ].bmRcvToggle = 0;

        epInfo[ XBOX_OUTPUT_PIPE_6 ].epAddr = 0x06; // XBOX 360 output endpoint - poll interval 8ms
        epInfo[ XBOX_OUTPUT_PIPE_6 ].epAttribs = USB_TRANSFER_TYPE_INTERRUPT;
        epInfo[ XBOX_OUTPUT_PIPE_6 ].bmNakPower = USB_NAK_NOWAIT; // Only poll once for interrupt endpoints
        epInfo[ XBOX_OUTPUT_PIPE_6 ].maxPktSize = EP_MAXPKTSIZE;
        epInfo[ XBOX_OUTPUT_PIPE_6 ].bmSndToggle = 0;
        epInfo[ XBOX_OUTPUT_PIPE_6 ].bmRcvToggle = 0;

        epInfo[ XBOX_OUTPUT_PIPE_7 ].epAddr = 0x07; // XBOX 360 output endpoint - poll interval 8ms
        epInfo[ XBOX_OUTPUT_PIPE_7 ].epAttribs = USB_TRANSFER_TYPE_INTERRUPT;
        epInfo[ XBOX_OUTPUT_PIPE_7 ].bmNakPower = USB_NAK_NOWAIT; // Only poll once for interrupt endpoints
        epInfo[ XBOX_OUTPUT_PIPE_7].maxPktSize = EP_MAXPKTSIZE;
        epInfo[ XBOX_OUTPUT_PIPE_7 ].bmSndToggle = 0;
        epInfo[ XBOX_OUTPUT_PIPE_7 ].bmRcvToggle = 0;
        
        epInfo[ XBOX_OUTPUT_PIPE_8 ].epAddr = 0x08; // XBOX 360 output endpoint - poll interval 8ms
        epInfo[ XBOX_OUTPUT_PIPE_8 ].epAttribs = USB_TRANSFER_TYPE_INTERRUPT;
        epInfo[ XBOX_OUTPUT_PIPE_8 ].bmNakPower = USB_NAK_NOWAIT; // Only poll once for interrupt endpoints
        epInfo[ XBOX_OUTPUT_PIPE_8].maxPktSize = EP_MAXPKTSIZE;
        epInfo[ XBOX_OUTPUT_PIPE_8 ].bmSndToggle = 0;
        epInfo[ XBOX_OUTPUT_PIPE_8 ].bmRcvToggle = 0;
        
        epInfo[ XBOX_OUTPUT_PIPE_9 ].epAddr = 0x09; // XBOX 360 output endpoint - poll interval 8ms
        epInfo[ XBOX_OUTPUT_PIPE_9 ].epAttribs = USB_TRANSFER_TYPE_INTERRUPT;
        epInfo[ XBOX_OUTPUT_PIPE_9 ].bmNakPower = USB_NAK_NOWAIT; // Only poll once for interrupt endpoints
        epInfo[ XBOX_OUTPUT_PIPE_9 ].maxPktSize = EP_MAXPKTSIZE;
        epInfo[ XBOX_OUTPUT_PIPE_9 ].bmSndToggle = 0;
        epInfo[ XBOX_OUTPUT_PIPE_9 ].bmRcvToggle = 0;

xxxajk, 2018-05-01:
Just an additional comment here, but... 

I can't wait for you (ahem... @Lauszus) to start porting to UHS3.0, because you won't have to worry so much about the parse problems with endpoints, etc, as the driver is provided with the information, and is interface, not "whole device" based.
It is much less confusing to write a driver for 3.0 once you understand that you don't have to do as much as is done in UHS2.0
so... @Lauszus get some free time already! Play hookie from work for a day.

What I hope to do quite soon is that basic joystick common interface, modeled as we discussed before, so that you just select what interfaces, and you get a common set of callbacks.

Soultaker, 2018-05-02:
@Lauszus Is it weird that there are only 2 in endpoints and 6 out endpoints and with non of them being at address 1 to 3?

xxxajk, 2018-05-03:
Considering I wrote the USB mass storage part, I would say "yes".
However that driver is very complex.
I recommend looking at the cdc-acm or one of the hid drivers, as they are the easiest to understand.

see https://github.com/felis/USB_Host_Shield_2.0/blob/master/cdcacm.cpp#L280
and https://github.com/felis/USB_Host_Shield_2.0/blob/master/cdcacm.cpp#L288

That is all there is to do what you want.

Soultaker, 2018-05-03:
@xxxajk Thanks. Last thing regarding the endpoint descriptors for the xbox one wireless adapter.
Is it normal for none of the endpoints not to be at address 0x01 or that there are 2 IN endpoints while there are 6 OUT endpoints?

01

Device descriptor:
Descriptor Length: 12
Descriptor type: 01
USB version: 0201
Device class: 00
Device Subclass: 00
Device Protocol: 00
Max.packet size: 40
Vendor ID: 045E
Product ID: 02E6
Revision ID: 0100
Mfg.string index: 01
Prod.string index: 02
Serial number index: 03
Number of conf.: 01

Configuration descriptor:
Total length: 004A
Num.intf: 01
Conf.value: 01
Conf.string: 02
Attr.: A0
Max.pwr: FA

Interface descriptor:
Intf.number: 00
Alt.: 00
Endpoints: 08
Intf. Class: FF
Intf. Subclass: FF
Intf. Protocol: FF
Intf.string: 02

Endpoint descriptor:
Endpoint address: 84
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Endpoint descriptor:
Endpoint address: 85
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Endpoint descriptor:
Endpoint address: 08
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Endpoint descriptor:
Endpoint address: 04
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Endpoint descriptor:
Endpoint address: 05
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Endpoint descriptor:
Endpoint address: 06
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Endpoint descriptor:
Endpoint address: 07
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Endpoint descriptor:
Endpoint address: 09
Attr.: 02
Max.pkt size: 0040
Polling interval: 00

Addr:1(0.0.1)

xxxajk, 2018-05-03:
Yes.
Sometimes the endpoints can be dual-role as well, where the endpoint is bidirectional, endpoint 0 is.
Let's take a look at a pretend USB device, that contains two interfaces.

```
device (address 0x01)--+-- endpoint 0x0 (always bidirectional, always present)
                       |
                       +-- interface 0 --+-- endpoint 0x01
                       |                 +-- endpoint 0x82
                       |
                       +-- interface 1 --+-- endpoint 0x03
                                         +-- endpoint 0x83
```
To get an idea on how to map this in your head, use the following formula.
{Device Address}:{Interface address}:{endpoint address}
...so, to talk to the device **as a whole**, you use:

```
0x01:0x00:0x00
`

And to talk to a specific **interface**, you would think about it like:
`
Interface 1
0x01:0x01:0x01
and
0x01:0x01:0x02

Interface 2
0x01:0x02:0x03
```

Please note:
Each endpoint has a specific role, and that role is in the endpoint descriptor. Some may only have read endpoints, or only write endpoints, or some other combination. Even no endpoints is totally legal.

Furthermore, interface  AND endpoint addresses **MAY APPEAR IN ANY ORDER and MAY EVEN SKIP**!

For example:
```
device (address 0x01)--+-- endpoint 0x0 (always bidirectional, always present)
                       |
                       +-- interface 3 --+-- endpoint 0x05
                       |                 +-- endpoint 0x83
                       |
                       +-- interface 2 --+-- endpoint 0x01
                                         +-- endpoint 0x81
```

This is why you have to **PARSE** the interface descriptors if you want to do things in a pedantic way.
**Unless you are 100% certain that none of the addresses ever changes, parse the descriptors!**

Note also your interface class/subclass and protocol that you posted:
```
Intf. Class: FF
Intf. Subclass: FF
Intf. Protocol: FF
```
This means that the interface is totally vendor specific, and does not follow any published USB standard. Please read more at the URL that I gave to you previously.

xxxajk, 2018-05-03:
Updated the comment to be more correct.

Soultaker, 2018-05-03:
Thank you so much

Mixinu, 2019-06-29:
Hello,

Did you manage to make the XBOXRECV code from UHS 2.0 library work with the Xbox One USB wireless dongle?

Lauszus, 2019-06-30:
@Mixinu no, as never got the hardware. I'm working on adding support for Xbox One S controllers via Bluetooth. Please test it if you have one: https://github.com/felis/USB_Host_Shield_2.0/pull/471.

hppyrbt, 2020-05-02:
Hello!
I am very interested and would love to have support for the Xbox One dongle in the USB host shield library.
I could help by donating a dongle for the development of the project, since I do not have enough knowledge to add the support on my own.

Is it possible?
Please, please.

sassacontrol, 2023-12-31:
@Lauszus Is it possible to use this Linux driver as a reference to use the Xbox-One Wireless Adapter on USB Shield?

[](https://github.com/medusalix/xone)

https://github.com/medusalix/xone

I saw that it is necessary to download Firmware FW_ACC_00U.bin

src-dev, 2024-03-26:
I am also willing to send hardware and make a donation if this can be implemented.