felis/USB_Host_Shield_2.0 #358: Xbox 360 Controller Failing to Connect

Treeaza, 2018-03-06:
I'm using a Sparkfun USB host shield (https://www.sparkfun.com/products/9947) connected to an Uno and running the Xbox USB example sketch. Previously this worked, but I was unable to get it to function reliably, and now it doesn't seem to recognize the Xbox controller at all. The Arduino connects to the USB shield correctly, but then (based on some Serial.println() statements I added to the loop function) never connects to the controller. The controller's LEDs initially have 1 and 4 on, then go dark, then begin blinking the entire ring. This may not be an issue with how I am using the library (if it isn't I apologize for wasting your time) but the hardware all seems to be working correctly. I have tested two seperate Unos, and made sure the controller is still working through my PC, so I'm not sure what else could be wrong at this point.

Lauszus, 2018-03-06:
Please connect pin 7 to reset on the Arduino and connect an external power supply.

Also please enable serial debugging, so I can see what is going on. If that does not work, then please share your code here in the comments.

Treeaza, 2018-03-06:
Thanks so much for the quick response. Connecting pin 7 to reset made it work immediately. Everything seems to be functioning exactly as intended now. Thank you!

Lauszus, 2018-03-06:
Great to hear that solved the problem.

closed 2018-03-06 by Lauszus