felis/USB_Host_Shield_2.0 #359: Interface with generic USB wireless m-bus dongle

marcogovo, 2018-03-06:
Dear Sirs,

I would interface my Arduino with the following wireless m-bus USB-adapter, [link](http://shop.imst.de/wireless-modules/usb-radio-products/10/im871a-usb-wireless-m-bus-usb-adapter-868-mhz?c=7)

We usually interface with that device through our Linux machine, using python and sending commands
to the HCI that communicates through serial.

But now,  we would try to do the same thing but using the Arduino.
With your library is possibile right now?
Or we have to write from scratch our libraries?

Thank you so much,
Good work,

Best Regards,
Marco

Lauszus, 2018-03-06:
Try the following example: <https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/acm/acm_terminal>. 

If that doesn't work then run the following example and post the output: <https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/USB_desc>.

marcogovo, 2018-03-07:
Dear Lauszus,

- We executed the acm_terminal.ino, removing the _/* reading the keyboard */_ code, 
because we don't care to read from our serial; we leave only the code that receives data from acm terminal.
But, we don't see anything on the serial monitor, even when the USB stick should receive data.
We are sure that the USB stick returns something on the serial, because when we use it with Python, for example, we correct see the data received.

So, i suppose we still not read correctly from the USB stick..

- At second, we executed the USB_desc.ino, and it returns the following results:

___________________________________________________________________

Start

01
—

Device descriptor: 
Descriptor Length:  12
Descriptor type:  01
USB version:    0200
Device class:    00
Device Subclass:  00
Device Protocol:  00
Max.packet size:  40
Vendor  ID:    10C4
Product ID:    EA60
Revision ID:    0100
Mfg.string index:  01
Prod.string index:  02
Serial number index:  03
Number of conf.:  01

Configuration descriptor:
Total length:    0020
Num.intf:    01
Conf.value:    01
Conf.string:    00
Attr.:      80
Max.pwr:    32

Interface descriptor:
Intf.number:    00
Alt.:      00
Endpoints:    02
Intf. Class:    FF
Intf. Subclass:    00
Intf. Protocol:    00
Intf.string:    02

Endpoint descriptor:
Endpoint address:  01
Attr.:      02
Max.pkt size:    0040
Polling interval:  00

Endpoint descriptor:
Endpoint address:  82
Attr.:      02
Max.pkt size:    0040
Polling interval:  00

Addr:1(0.0.1)
_________________________________________________________________

Thank you for the support,

Regards

Lauszus, 2018-03-15:
@marcogovo it does not seem to be a standard serial device. My recommendation would be to see if you can find a driver in the Linux kernel and then port it to the library.

marcogovo, 2018-03-16:
My USB Dongle is equipped with a CP210x serial adapter, 
so i found the driver in drivers/usb/serial/cp210x.c.

The problem now is how to port this library to arduino,
always if it is possible to do..
Do you suggest something?

The other way, maybe easier than the previous one, is to interface directly with the 
module inside the USB Dongle, using simply UART between them..
In this way i can bypass the USB communication avoiding the driver problems..

Thank you very much, 
Regards,

Marco

Lauszus, 2019-01-06:
Sorry for never getting back to you @marcogovo, but I simply do not have the time to write custom drivers for people.

closed 2019-01-06 by Lauszus