felis/USB_Host_Shield_2.0 #360: No matching function for call to "min"

VanKurt, 2018-03-07:
I can't get any example code to compile with Arduino IDE or PlatformIO (Visual Studio Code). The error message is always the same:

> .../lib/USB-Host-Shield-20_ID59/PS4Parser.cpp: In member function 'void PS4Parser::Parse(uint8_t, uint8_t*)':
> .../lib/USB-Host-Shield-20_ID59/PS4Parser.cpp:87:105: **error: no matching function for call to 'min(uint8_t, unsigned int)'**
> memcpy(&ps4Data, buf + 1, min((uint8_t)(len - 1), MFK_CASTUINT8T sizeof(ps4Data)));
> ^
> .../lib/USB-Host-Shield-20_ID59/PS4Parser.cpp:87:105: note: candidates are:
> ...

PS: I just downloaded the latest verson of this library and both IDEs. So everything should be up to date.

Lauszus, 2018-03-07:
What board are you using? The min function is defined here in the Arduino source code: https://github.com/arduino/Arduino/blob/a2a17a0c83afdd6812664f32161db0ccdb535ad6/hardware/arduino/avr/cores/arduino/Arduino.h#L92.

Also please include the library using the Arduino library manager: https://github.com/felis/USB_Host_Shield_2.0#arduino-library-manager.

We use Travis to compile the library every time a new commit is pushed, so it must be a problem on your side.

VanKurt, 2018-03-07:
I'm using an ESP8266 (Wemos D1 Mini). The library was included using the Arduino manager.

Hmm, strange :-/

Lauszus, 2018-03-07:
I just cleared the Travis cache and triggered a new build - let's see if that builds successfully: https://travis-ci.org/felis/USB_Host_Shield_2.0/builds/350503082.

Lauszus, 2018-03-15:
@VanKurt sorry I should have read your issue more carefully!

The problem is that the hardware add-on is not setting the defining ```ESP8266```, as this is a known issue with the ESP8266: https://github.com/felis/USB_Host_Shield_2.0/blob/a017f0f31159df7eaa4ca21fb6de75bf02553638/settings.h#L159-L161.

Can you try to compile the code again with verbose compiling enabled in the Arduino IDE and post the output? (Click on the checkbox in the setting labeled "Show verbose output during: Compiling").

Lauszus, 2019-01-06:
Closing due to inactivity.

closed 2019-01-06 by Lauszus