felis/USB_Host_Shield_2.0 #362: Ps4 controller serial problem on power

srknsntrk1991, 2018-03-13:
Hello 

I am trying to run a dc motor and ps4 controller which is connected to arduino uno r3 via usb cable. 

I debugged your example on my arduino and i run serial port screen and everything works well  and i can even arrange motor speed but when i remove my arduino from my computer and put it on power, ps4 controller doesnt work. 

I think there is a serial port problem about starting it. I would be very thankful if you could help me. 

Regards,
Serkan

Lauszus, 2018-03-13:
Simply remove the following line:

```C++
while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
```

closed 2018-03-13 by Lauszus

srknsntrk1991, 2018-03-13:
Ahah yes i did it before you say :P Thank you so much