felis/USB_Host_Shield_2.0 #363: Xbox One Rumble Support

Soultaker, 2018-03-13:
Hi all,

I have been going through the library for the Xbox One controller and wanted to ask if code was already written to enable rumble. I see that there is code in the Xbox USB library and tried to replicate that in the Xbox One library to no avail. If there is support already available, how can I code it such that holding any of the trigger button will rumble the Xbox One controller? Any help would be great.

Thanks!

Soultaker, 2018-03-15:
This is for a project that I am doing and am using the shield from here.

Lauszus, 2018-03-15:
@Soultaker please test #365 and report back. You can download the code at the following link: https://github.com/felis/USB_Host_Shield_2.0/archive/xboxone.zip and then open the XBOXONE.ino example :)

Soultaker, 2018-03-15:
@Lauszus Thank you for your help. Unfortunately I could not get the Xbox One controller to rumble with the current fix. The controller is an official Xbox One controller and one of the newer ones at that. Besides rumble, everything else is working as expected from the example code.

Regarding the code from the fix, this makes sense from my understanding and should have worked. I also did something similar but not  as robust as your code. Could it be that the rumble protocol has been changed?  Were you able to test this yourself  on your own controller? I ask only to cross out the possibility that the issue is a hardware one on my end.

Again thanks. :)

Lauszus, 2018-03-15:
@Soultaker the problem is that I don't have a controller myself, so I can not test it.

Does the controller rumble when it is first connected, as that has been working in the past?

Soultaker, 2018-03-15:
@Lauszus Yes the controller does rumble when it is first connected. But I believe that the controller performs this rumble on start up not because of the code to make it buzz but because the controller itself was made to buzz when it receives power. The reason I believe this is because I tried commenting out the writeBuf part of onInit and the controller still buzzes on start up.

Lauszus, 2018-03-15:
Okay. Does it rumble when you plug it into a computer?

Are you 100 % sure that you are actually modifying and compiling the source code? Please follow the manual installation instructions if you have not already: https://github.com/felis/USB_Host_Shield_2.0#manual-installation.

The reason I am asking is because I have had other people in the past saying they were modifying the code, but they really wasn't...

Can you make a clear error in the code, so the compiler fails?

Soultaker, 2018-03-15:
@Lauszus When I make a change to the example code, is it supposed to say some files are marked as read only?

Lauszus, 2018-03-15:
Yes that is how Arduino works. Just edit the files using a text editor and set the Arduino IDE to use an external editor: http://wiki.seeed.cc/Use_External_Editor/.

Soultaker, 2018-03-15:
So yes I can make a clear error and yes I am actually modifying and compiling the source code. Apologies for the late response, I went back to double check ergarding your suggestion for for manual installation.

Soultaker, 2018-03-15:
The clear error I made was a simple removal of ";" from a line of code

Lauszus, 2018-03-15:
Okay. Did you also try modifying the actual source of the library? Please confirm that the additions in the pull request are present.

Soultaker, 2018-03-15:
A check that I made to see if it goes through the code on the example code was to print hello world on the serial monitor. I placed this right below the setRumbleOn  in the if else statement that checks for (leftRumble > 0 || rightRumble > 0).

Lauszus, 2018-03-15:
Please try to edit source code of the library.

Soultaker, 2018-03-15:
@Lauszus So yes I did try modifying the actual source of the library. By enabling serial debugging in serial.h, I had serial output "love" after placing the following line of code "Notify(PSTR("\r\nLove: "), 0x80);" right before "XboxCommand(writeBuf, 12);" in the setRumbleOn function of the XboxOne.cpp.

Soultaker, 2018-03-15:
So the serial monitor does output "love" whenever I hold down the right trigger.

Lauszus, 2018-03-15:
Okay thanks. I just wanted to make sure that you you were using the right code.

Anyway there is not much I can do then, as I do not have the controller.

Soultaker, 2018-03-15:
Is it possible that the reason is the XboxCommand uses pUsb->outTransfer instead of pUsb->ctrlReq like in XBOXUSB and XBOXRECV?

Soultaker, 2018-03-15:
@Lauszus Also are you still able to contact guruthree who initially created this support? I apologize for taking up so much of your time and thank you for your help.

Soultaker, 2018-03-15:
Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	47
Device Protocol:	D0
Max.packet size:	40
Vendor  ID:		045E
Product ID:		02D1
Revision ID:		0203
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		0060
Num.intf:		03
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	01
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0040
Polling interval:	04

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	02
Attr.:			01
Max.pkt size:		00E4
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			01
Max.pkt size:		00E4
Polling interval:	01

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		47
Intf. Protocol:		D0
Intf.string:		00

Endpoint descriptor:
Endpoint address:	03
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Endpoint descriptor:
Endpoint address:	83
Attr.:			02
Max.pkt size:		0040
Polling interval:	00

Addr:1(0.0.1)

Soultaker, 2018-03-15:
Is this what you need?

Lauszus, 2018-03-15:
Yes that is exactly what I needed :)

From the code I can actually see that you are using a slightly older revision of the Xbox One controller?

Can you try to revert this pull request: https://github.com/felis/USB_Host_Shield_2.0/pull/325/files and see if that works? Please try the following code: https://github.com/felis/USB_Host_Shield_2.0/archive/1.3.1.zip.

Soultaker, 2018-03-15:
Does it matter if I am working on a linux ubuntu machine?

xxxajk, 2018-03-15:
Nope.

Lauszus, 2018-03-15:
No, that does not make any difference :)

Soultaker, 2018-03-15:
So using the above code, I did not see any changes compared to the latest code.

Soultaker, 2018-03-15:
I'll look through the web to see if there are any new protocols

Lauszus, 2018-03-15:
I just checked the Linux kernel and the command they use is slightly different. I will push commit in a bit.

Lauszus, 2018-03-15:
@Soultaker I just pushed a commit. Please download the code and report back: https://github.com/felis/USB_Host_Shield_2.0/archive/xboxone.zip.

Soultaker, 2018-03-15:
@Lauszus I am happy to report that rumble is now possible on the controller

Lauszus, 2018-03-15:
Great! :D Does it still rumble briefly when it is first connected?

Lauszus, 2018-03-15:
@Soultaker I made a small change. Can you please download the code again and test it?

Soultaker, 2018-03-15:
Rumble strength from the triggers is static no matter how far I pull the triggers, so I will try experimenting to get the rumble strength to correspond to the triggers. 

Again thank you! :)

Soultaker, 2018-03-15:
I will try again with the second download right now

Soultaker, 2018-03-15:
@Lauszus Do I use the same link as above?

Lauszus, 2018-03-15:
You're welcome! Yes, just use the same link, as it will download the code from the branch I use for the PR.

Lauszus, 2018-03-15:
You can try to modify the report, so it uses the "Continous rumble effect" report as described here: https://github.com/quantus/xbox-one-controller-protocol#0x09-activate-rumble.

Lauszus, 2018-03-15:
Please also verify that the triggers return a unsigned 16-bit number i.e. the value should go up to 65535?

Soultaker, 2018-03-15:
@Lauszus So iteration of the code also works. Triggers will start their respective motors at a static strength.

Soultaker, 2018-03-15:
So the triggers return a value that goes as high as 1023

Lauszus, 2018-03-15:
I just pushed a fix. Please test again.

Soultaker, 2018-03-15:
I think you should change the value to 2047. With 1023, the motors rotate so fast that they feel like they may break if the triggers held for a long time.

Lauszus, 2018-03-15:
No if that is the maximum the controller support, then lets just use that, as I doubt Microsoft would make anything that could break itself.

Soultaker, 2018-03-15:
Okay then. Just some concerns since the feedback is really strong at 1023 and with the original 65535 the feedback was weak.

Lauszus, 2018-03-15:
Note that I simply use 1023 to map the value from 0-1023 to 0-255. It's just meant as an example, so you can always change it to whatever you prefer.

Soultaker, 2018-03-15:
Thanks! This really helps!

Soultaker, 2018-03-15:
There is a strange behavior where disconnecting and reconnecting the controller causes controller inputs to not be read anymore until the serial monitor is restarted. I will deal with this on my end since this would be considered an edge case but I just wanted to let you know.

closed 2018-03-15 by Lauszus

Lauszus, 2018-03-15:
@Soultaker I just released a new version of the library, so rumble support should be available via the Arduino library manager in 24hr or so :)

What board are you using? The library should not be affected by the serial communication between the board and your computer, so it sounds more like a problem with the ```Serial``` implementation.

Are you resetting the board by any change, as then the board will wait at this line if it is using a built-in USB CDC serial connection:

```C++
while (!Serial);
```

Soultaker, 2018-03-15:
The board i am using is an Arduino UNO

Lauszus, 2018-03-16:
@Soultaker okay. Can you try to blink a LED or something to check if the board truly hangs? Note that you can not use the built-in LED as it is connected to pin 13, which is used by the shield for SPI communication.

Lauszus, 2018-03-16:
Note that when you restart the serial monitor you are also resetting the Arduino, so that might be the reason why it works.

Can you try to mount a power switch and see if it makes any difference: https://www.circuitsathome.com/mcu/vbus-power-control-on-usb-host-shield/?

Soultaker, 2018-03-17:
@Lauszus So I won't be able to test these things until Monday

Soultaker, 2018-03-17:
You are right in that I am resetting the arduino to get things to work after removing the controller from the USB shield. There is also the fact that when I plug in a new controller, the board is able to pick up on it and start reading inputs for that controller.

Soultaker, 2018-03-17:
Possible that the board is not reading the disconnect

Soultaker, 2018-03-23:
@Lauszus So the problem was not the library itself but how I implemented the code in the example.

PR #365 (2018-03-15): Added support for rumble effect on Xbox One controllers
Fixes #363
commit b0b255cd (2018-03-15): Added support for rumble effect on Xbox One controllers

Fixes #363