felis/USB_Host_Shield_2.0 #364: GPLv3 Compatibility?

marcio-ao, 2018-03-13:
Hello,

We would like to incorporate the code from the USB Host Shield in the [Marlin firmware](http://marlinfw.org/) for 3D printers. However, I notice that the USB Host Shield is released under "GPL v2" rather than "GPLv2 or later".

Is there any chance the license could be upgraded to either the GPL v3, or to include the wording "GPLv2 or later" in order to maintain compatibility with the GPL v3?

Information why this is necessary is GPLv3 and GPLv2 listed here:

https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility

Thank you!

-- Marcio

felis, 2018-03-14:
I like Lulzbot and I like my TAZ 6 :-), the whole product documentation for which is open sourced and available.

@Lauszus , @xxxajk  - do you have objections to changing the wording of the license to 'GPLv2 or later'?

Lauszus, 2018-03-15:
I'm in for changing the license as well. I have a few 3D printers running Marlin, så I'm excited to see what they come up with :)

xxxajk, 2018-03-15:
I've no problem with GPL2 or later...

marcio-ao, 2018-03-15:
This is fantastic news! Another engineer and I have already done preliminary work on this and we know that we can reuse the FAT routines that Marlin already has for SD cards, so we only need the following files that implement the block-level USB mass storage access:

- address.h
- avrpin.h
- confdescparser.h
- macros.h
- masstorage.cpp
- masstorage.h
- max3421e.h
- message.cpp
- message.h
- parsetools.cpp
- parsetools.h
- printhex.h
- settings.h
- Usb.cpp
- Usb.h
- usb_ch9.h
- UsbCore.h
- usbhost.h

So for our purposes, if just those particular files were updated with the wording "GPL v2 or later", we would be able to perform the integration.

Thank you so much!

-- Marcio

marcio-ao, 2018-03-19:
I just wanted to share some preliminary results. This is actually Marlin reading files from a USB drive. It is also driving a color LCD touch panel (another project we have been working on):

[image]

Printing already works, although the only glitch is that currently the drive insertion/removal is not recognized.

xxxajk, 2018-03-19:
Very cool.
Thanks for sharing!

marcio-ao, 2018-03-20:
Curious whether there will be a commit with the files with an updated license? This is not something I feel comfortable doing myself. I would like to replace the files I am currently using with something with the correct licenses.

Lauszus, 2018-03-20:
Since @felis assigned himself, I was assuming that he would do it?

marcio-ao, 2018-04-04:
Anything happening with this?

Lauszus, 2018-04-10:
@marcio-ao I have updated the copyright header in the relevant files in https://github.com/felis/USB_Host_Shield_2.0/commit/00aa35518b618e75af5cee29297c1abfb0af969b.

closed 2018-04-10 by Lauszus

marcio-ao, 2018-04-10:
@Lauszus : Awesome! Thank you so much!

Lauszus, 2018-04-10:
You're welcome :)

marcio-ao, 2018-05-08:
@Lauszus : Sorry to bother you again, is there a chance you could update the license for the following files as well?

- version_helper.h
- hex_dump.h
- sink_parser.h

I'm sorry I didn't request this sooner. I got caught up in a new printer releases and I had to put this on hold for a while.

Lauszus, 2018-06-23:
@marcio-ao sorry for the late reply, it was fixed in: https://github.com/felis/USB_Host_Shield_2.0/commit/06d5ed134a37e22575c0ce18c061d9ef115151e0.

marcio-ao, 2018-06-25:
@Lauszus : Thank you! I was able to roll it all up as a PR to Marlin (#11115). Hopefully we will see this new feature standard in Marlin 2.0. I appreciate all the help along the way! :)

Lauszus, 2018-06-25:
Great! Looking forward to seeing the code in the wild :)

commit 00aa3551 (2018-04-10): Release part of the code under "GPLv2 or later", so it is compatible with GPLv3

Fixes #364