felis/USB_Host_Shield_2.0 #367: Access to SDA & SCL pins

Soultaker, 2018-03-23:
@Lauszus Hi! I'm trying to connect a IIC/I2C Serial Interface Adapter Module for Arduino on the USB host shield and was wondering if it was possible to access the SDA and SCL pins that have been covered up on the Arduino UNO. I know that there is already support for LCD screens and this is more for convenience and curiosity if anything else.

Thanks!

Lauszus, 2018-03-23:
The I2C interface is available on pin A4 and A5 on the Uno as well.

Soultaker, 2018-03-23:
Apologies. This was a misunderstanding on my part. I can still use the A5 and A4 pins even with the Host shield on. You can close this.

Lauszus, 2018-03-24:
No worries, I'll close this issue then. Note that you can close issues yourselves if you opened them.

closed 2018-03-24 by Lauszus