felis/USB_Host_Shield_2.0 #369: Mouse disconnected and no debug output

kevingilboy, 2018-03-27:
Hi Felis, I am trying to connect to a simple USB mouse using a USB host mini, Arduino Uno R3, and the example sketch _examples>HID>USBHIDBootMouse_. 

Using some Serial.print() debugging, I found that the program makes it past `Usb.Init()` (which returns 0 for valid initialization) but does not report any data from the mouse. Looking into the `Usb.Task()` method that is performed every loop, I see that the `tmpdata` variable is set to `SE0` which is a disconnected state.

I would provide debug output but unfortunately setting the `ENABLE_UHS_DEBUGGING` flag to 1 in settings.h does not appear to produce any serial output...

Do you have any thoughts on either why debug output is not being produced or why the USB task determines that the mouse is disconnected?

Much appreciated :)

Lauszus, 2018-03-27:
Have you tried to cut the VBUS jumper and connected 5V to the VBUS pad? Please see the hardware manual for instructions: https://www.circuitsathome.com/usb-host-shield-hardware-manual/.

kevingilboy, 2018-03-27:
Hi Lauszus, thank you for the suggestion. Unfortunately I do have it set up that way :(

xxxajk, 2018-03-28:
Uno is 3.3v? If not, you might have destroyed the UHS mini...

Lauszus, 2019-01-06:
Closing due to inactivity.

closed 2019-01-06 by Lauszus