felis/USB_Host_Shield_2.0 #370: How can you transmit audio using a Bluetooth Dongle?

MCASEY200, 2018-03-27:
Hey Guys,

I’m just a beginner with Arduino and what I’m trying to do is take an incoming audio signal and then transmitted through a Bluetooth dongle. The Bluetooth dongle that I’m using for this is the Sennheiser BTD 800 USB Bluetooth Network Card. What I’m wanting to do is to display the incoming audio signal in the plotter window.  I’ve currently been messing about with this SPP sketch from the USB Host shield library 2.0. Which you can see below. If anyone has any ideas on how to program it to do this, If anyone knows how to code this I would really appreciate this, Thanks.

`#include
#include

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include
#endif
#include

USB Usb;
// USBHub Hub1(&Usb); // Some dongles have a hub inside

BTD Btd(&Usb); // You have to create the Bluetooth Dongle instance like so
/* You can create the instance of the class in two ways */
SPP SerialBT(&Btd); // This will set the name to the defaults: “Arduino” and the pin to “0000”
int incomingAudio = A0;

void setup() {
Serial.begin(9600);
#if !defined(__MIPSEL__)
while (!Serial); // Wait for serial port to connect
// – used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
if (Usb.Init() == -1) {
Serial.print(F(“\r\nOSC did not start”));
while (1); //halt
pinMode(incomingAudio, INPUT); // set pin A0 to be input

}
}

void loop() {

Usb.Task(); // The SPP data is actually not send until this is called, one could call SerialBT.send() directly as well

(SerialBT.connected); {
incomingAudio = analogRead(A0);
SerialBT.println((incomingAudio)); // Send INCOMING AUDIO
}
{
if (Serial.available())
SerialBT.write(Serial.read());
if (SerialBT.available())
Serial.write(SerialBT.read());
}
}`

Lauszus, 2018-03-28:
@MCASEY200 it looks like you have messed up the if-statements? Please use Markdown to highlight your code correctly: https://help.github.com/articles/creating-and-highlighting-code-blocks. I have trying to edit you existing code, but it is unclear to me if the if-statement ```(SerialBT.connected); {``` was messed up because of how the Github comments works or not.

I'm not sure I understand 100 % what you are trying to do? If you are trying to stream audio, then I am afraid that that is not supported by the library, as the Bluetooth audio protocol (A2DP) is not implemented.

MCASEY200, 2018-03-29:
No that was me just messing about with the sketch. What I was trying to do was take an audio source that was coming in to pin A0 on my Arduino Due, and then I thought I could transmit that through the  Bluetooth dongle. I thought it was just as simple as pairing the Arduino what a device and then sending the data, but thanks for the help  tho man.

Lauszus, 2018-04-12:
@MCASEY200 yes you can send the data via SPP, as it acts just like a serial port.

Lauszus, 2019-01-06:
Closing due to inactivity.

closed 2019-01-06 by Lauszus