felis/USB_Host_Shield_2.0 #373: Issue with xbox 360 wireless receiver causing output pins to be constant high output

jhcumby, 2018-04-03:
We are using an arduino uno microcontroller with a generic xbox 360 wireless receiver dongle. We are using the XBOXRECV source code to use an xbox 360 controller to controler DC motor controllers and a 5V one channel relay. As soon as the wireless dongle is connected, pins 10 through 13 begin outputting a slight voltage signal. Even when the pins 10, 11, 12, and 13 are defined as outputs and set to be a LOW signal, an above 0 voltage signal is detected at those pins.

Lauszus, 2018-04-04:
Those pins are used for the SPI communication, please see: https://github.com/felis/USB_Host_Shield_2.0#interface-modifications.

closed 2018-04-04 by Lauszus

jhcumby, 2018-04-04:
Thanks, that clears up my issue!

jhcumby, 2018-04-04:
Would it be possible to still send a pwm signal from pin 9 to my motor controller even though pin 9 is used as INT pin?