felis/USB_Host_Shield_2.0 #374: Control transfer

jef737, 2018-04-06:
Hi,
I need to receive data from a device which use control transfer protocol. Then if i well understood i need to send an Usb.ctrlReq to initiate the control transfer. In despite of many search on the net i didn't find any documentation or example about how to use the ctrlReq. I tried with something like this: rcode=Usb.ctrlReq(address, endpoint, numberofbyte, data) but i received the return  code:0xDB. 
All advices will be welcome.

jef737, 2018-04-07:
Hi,
Update. In fact i didn't use the endpoint 0 to send the control transfer, then i changed with the right endpoint and now i get the return code:5. Doe's someone know what this return code means? An another point the Usb.ctrlREQ codification given above is not correct, bmrequesttype, brequest, wValue, wIndex was missing.

xxxajk, 2018-04-08:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/max3421e.h#L212

which basically means "try again later".

jef737, 2018-04-09:
Thanks for this information. 
Now the ctrlREQ is correct but now i get the same error return code 0xDB with Usb.inTransfer. It would be great if someone had a code example with control transfer because i can't go further.

Lauszus, 2018-04-09:
@jef737 the best "examples" you can get are by looking at the different drivers. For instance take a look at the PS3USB driver: https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3USB.cpp.

jef737, 2018-04-09:
Thanks @Lauszus 
I turn around and stay with the same return code 0xDB. here is an extract of my code:
         epInfo[0].epAddr = 0x00; // control endpoint 
          epInfo[0].epAttribs = USB_TRANSFER_TYPE_CONTROL;
          epInfo[0].maxPktSize = 8;
          epInfo[0].bmNakPower = USB_NAK_DEFAULT;
          epInfo[0].bmSndToggle = 0;
          epInfo[0].bmRcvToggle = 0;
          epInfo[1].epAddr = 0x81; // data endpoint
          epInfo[1].epAttribs = USB_TRANSFER_TYPE_INTERRUPT;
          epInfo[1].maxPktSize = 8;
          epInfo[1].bmNakPower = USB_NAK_DEFAULT;
          epInfo[1].bmSndToggle = 0;
          epInfo[1].bmRcvToggle = 0;

          rcode = Usb.setEpInfoEntry(1, 1, epInfo);
          Serial.println("setEpInfoEntry:");
          Serial.println(rcode, HEX);
          
          rcode = Usb.ctrlReq( 1, epInfo[0].epAddr, 0x21, 0x09, 0x00, 0x02 ,0x0000, 1, 8, CMD, NULL);
          Serial.println("ctrReq:");
          Serial.println(rcode, HEX);  

          rcode = Usb.inTransfer(1, epInfo[1].epAddr, &nbByter, data); 
          Serial.println("inTransfer:");
          Serial.println(rcode, HEX); 
All return code up to the last one are correct. 
Could you tell me what is wrong in my code?

jef737, 2018-04-10:
Update: I've changed the setEpInfoEntry like this: Usb.setEpInfoEntry(1, 2, epInfo) 
Now i get the return code 0x05 which means STALL handshake. Could you tell me how the usb library works with the control transfer. I wrote a program in C with the libusb library which works fine, in first i send to the device a control transfer packet with 8 bytes of Operation Command data, and just after i receive from the device the available data with an interrupt transfer function. From the USB protocol view the host send the control request packet, then the device send a control packet following by a Set configuration status packet, finally the device send the available data. 
Therefore with the libusb library all the usb control transfer protocol is managed with only 2 instructions one libusb_control_transfer and one libusb_interrupt_transfer. 
Does the usbhostshield library works on the same way or is it necessary to manage all the control transfer steps?