felis/USB_Host_Shield_2.0 #376: Read from 4 HID Bar Code Scanners via hub

akaufman1, 2018-04-23:
Hey all,
I am trying to read HID keyboard data from 4 Zebra DS457 scanners.  Each scanner is connected to a StarTech industrial hub powered by a lab power supply.  Each scanner has a trigger wire connected to IO on the Arduino.  The Arduino is also powered.  

[image]

I am able to get scans from the first two scanners but the second two scanners won't trigger and won't output any data.  I setup each scanner the same way, and I have run USB_desc and I can get the descriptor info for all 4 scanners... 

Below is my code which is based on the USBHIDBootKbd example.  Any advice or thoughts appreciated!
__________________________________________________________________________________________________________

#include <hidboot.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

int curScan = 0;

const int SCAN0_TRIGGER_PIN = A0; // Scanner Trigger pin - active-low
const int SCAN1_TRIGGER_PIN = A1; // Scanner Trigger pin - active-low
const int SCAN2_TRIGGER_PIN = A2; // Scanner Trigger pin - active-low
const int SCAN3_TRIGGER_PIN = A3; // Scanner Trigger pin - active-low

class KbdRptParser : public KeyboardReportParser
{
    void OnKeyDown	(uint8_t mod, uint8_t key);
    void OnKeyPressed(uint8_t key);
};

void KbdRptParser::OnKeyDown(uint8_t mod, uint8_t key)
{
  uint8_t c = OemToAscii(mod, key);

  if (c)
    OnKeyPressed(c);
}

void KbdRptParser::OnKeyPressed(uint8_t key)
{
  if(key == 19){
    Serial.print(" - Scanner ");
    Serial.println(curScan);
  }
  else
    Serial.print((char)key);
};

USB     Usb;
USBHub  Hub(&Usb);
HIDBoot<USB_HID_PROTOCOL_KEYBOARD>    HidKeyboard0(&Usb);
HIDBoot<USB_HID_PROTOCOL_KEYBOARD>    HidKeyboard1(&Usb);
HIDBoot<USB_HID_PROTOCOL_KEYBOARD>    HidKeyboard2(&Usb);
HIDBoot<USB_HID_PROTOCOL_KEYBOARD>    HidKeyboard3(&Usb);

KbdRptParser Prs;

unsigned long scanTime = 0;

void setup()
{
  Serial.begin( 115200 );

  pinMode(SCAN0_TRIGGER_PIN, OUTPUT);
  pinMode(SCAN1_TRIGGER_PIN, OUTPUT);
  pinMode(SCAN2_TRIGGER_PIN, OUTPUT);
  pinMode(SCAN3_TRIGGER_PIN, OUTPUT);
  
  digitalWrite(SCAN0_TRIGGER_PIN, HIGH);
  digitalWrite(SCAN1_TRIGGER_PIN, HIGH);
  digitalWrite(SCAN2_TRIGGER_PIN, HIGH);  
  digitalWrite(SCAN3_TRIGGER_PIN, HIGH);
  
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
    Serial.println("OSC did not start.");

  delay( 200 );

  HidKeyboard0.SetReportParser(0, &Prs);
  HidKeyboard1.SetReportParser(0, &Prs);
  HidKeyboard2.SetReportParser(0, &Prs);
  HidKeyboard3.SetReportParser(0, &Prs);

  delay(5000);
  
  scanTime = millis();
}

void loop()
{
  if (scanTime + 2500 < millis())
  {
    scanTime = millis();
    
    if (curScan < 3)
      curScan++;
     else
      curScan = 0;

     Serial.print ("Scanning ");
     Serial.println (curScan);
  }
  
  switch (curScan){
    case 0:
      digitalWrite(SCAN1_TRIGGER_PIN, HIGH);
      digitalWrite(SCAN2_TRIGGER_PIN, HIGH);
      digitalWrite(SCAN3_TRIGGER_PIN, HIGH);
      
      digitalWrite(SCAN0_TRIGGER_PIN, LOW);
      break;
    case 1:
      digitalWrite(SCAN0_TRIGGER_PIN, HIGH);
      digitalWrite(SCAN2_TRIGGER_PIN, HIGH);
      digitalWrite(SCAN3_TRIGGER_PIN, HIGH);
      
      digitalWrite(SCAN1_TRIGGER_PIN, LOW);
      break;
    case 2:
      digitalWrite(SCAN0_TRIGGER_PIN, HIGH);
      digitalWrite(SCAN1_TRIGGER_PIN, HIGH);
      digitalWrite(SCAN3_TRIGGER_PIN, HIGH);
      
      digitalWrite(SCAN2_TRIGGER_PIN, LOW);
      break;
    case 3:
      digitalWrite(SCAN0_TRIGGER_PIN, HIGH);
      digitalWrite(SCAN1_TRIGGER_PIN, HIGH);
      digitalWrite(SCAN2_TRIGGER_PIN, HIGH);
      
      digitalWrite(SCAN3_TRIGGER_PIN, LOW);
      break;
  }
  
  Usb.Task();
}

Lauszus, 2018-04-23:
Try to create on more USBHub instance:

```C++
USBHub  Hub2(&Usb);
```

akaufman1, 2018-04-25:
The issue has something to do with a newer version the scanner module.  I purchased two more of the older model and thing work as expected.  I will dig into what is different in the configuration of the scanner... but this issue is closed.  Thanks!

closed 2018-04-25 by akaufman1