felis/USB_Host_Shield_2.0 #377: RFID/barecode

farahammar, 2018-04-24:
I'm looking for help:
I have a project with arduino uno + barecode reader +RFID module RC522.
I want to read this barecode in arduino and send it to the RFID tag . How can i do this?﻿

Lauszus, 2018-04-29:
Please see: https://github.com/felis/USB_Host_Shield_2.0/issues/286. There is an example for using a barcode scanner here: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/adk/adk_barcode/adk_barcode.ino.

farahammar, 2018-05-02:
i tried with this code but i have some problem:     
fatal error: hid.h: No such file or directory

 #include <hid.h>

cyberjupiter, 2018-05-08:
Your question is way too general. You should test each device independently first and make sure they work and there's no compilation error. For example, test the USB barcode scanner first. Only include the necessary library and output the barcode into the Serial Monitor so you know you are getting it right. 

Then, start combining them. However, do note that both devices use SPI protocol, so you have to read about it first. There are lots of information on the web.

Regarding your problem, check that the USB Host Shield library is properly installed in your Arduino library folder.

Lauszus, 2018-06-23:
@farahammar some excellent advice from @cyberjupiter above. To put it simple: do baby steps and then eventually you will get there.

I will close this issue now, as this is not the place to ask for coding guidance.

closed 2018-06-23 by Lauszus