felis/USB_Host_Shield_2.0 #378: Non-original XBox 360 gamepad error 0d

AJIOB, 2018-04-29:
Hello. I'm trying to connect `Defender Game Racer Turbo RS3` gamepad to USB Host Shield using your `XBoxUSB` library. But it was not detected. I turn on debug in `XBOXUSB.cpp` trying to find a problem using prints, but can only detect, that `USB::ctrlReq` method generates an error `0x0D` when calling `USB::dispatchPkt`. How can I fix this gamepad detection?

Error log (`AJIOB 2` and vid & pid prints are simple `Serial.print` calls):

```
XBOX USB Library Started
XBOXUSB Init
Vid 1118
Pid 654
Addr: 01
AJIOB 2

setConf 
Xbox 360 Init Failed, error code: 0D

XBOXUSB Init
AJIOB 2

getDevDescr 
Xbox 360 Init Failed, error code: 0D

XBOXUSB Init
AJIOB 2

getDevDescr 
Xbox 360 Init Failed, error code: 0D

XBOXUSB Init
AJIOB 2

getDevDescr 
Xbox 360 Init Failed, error code: 0D
```

I tested it on Arduino UNO R3.
P.S. Windows 10 detects this gamepad as `XBox 360 Controller for Windows`

Lauszus, 2018-04-29:
Please run the following example and post the output: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USB_desc/USB_desc.ino.

Btw what shield are you using? Also try to connect an external power supply.

AJIOB, 2018-04-29:
I'm getting such output:
```
Start

01
--

AJIOB 2

Request error. Error code:	0D

Addr:1(0.0.1)
```
I'm using shield `Keyes USB Host Shield` (visually it's a copy of mine shiled):
[image]
Sorry, but I have no external power supply, I cannot connect it.

AJIOB, 2018-04-29:
I'm trying to connect gamepad and then press "Reset" button. I'm getting such output:
```

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	FF
Device Protocol:	FF
Max.packet size:	08
Vendor  ID:		045E
Product ID:		028E
Revision ID:		0110
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0030
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			80
Max.pwr:		FA

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		5D
Intf. Protocol:		01
Intf.string:		00
Unknown descriptor:
Length:		10
Type:		21
Contents:	10010124811403000313020003000705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0020
Polling interval:	04

Endpoint descriptor:
Endpoint address:	02
Attr.:			03
Max.pkt size:		0020
Polling interval:	08

Addr:1(0.0.1)
```

Lauszus, 2018-04-29:
Please post the descriptor of the other controller as well, so I can compare the two.

Also the descriptor you just posted, is that for the original controller or the other one?

AJIOB, 2018-04-29:
It was a descriptor of Defender gamepad. I can publish an original descriptor only on next Friday

Lauszus, 2018-04-29:
Yes please do.

Kimentii, 2018-05-04:
Descriptor of original xbox 360 gamepad.
```
Start
 Protocol:              13
Intf.string:            04
Unknown descriptor:
Length:         06
Type:           41
Contents:       000101038000

Addr:1(0.0.1)
Start

01
--

Device descriptor:
Descriptor Length:      12
Descriptor type:        01
USB version:            0200
Device class:           FF
Device Subclass:        FF
Device Protocol:        FF
Max.packet size:        08
Vendor  ID:             045E
Product ID:             028E
Revision ID:            0114
Mfg.string index:       01
Prod.string index:      02
Serial number index:    03
Number of conf.:        01

Configuration descriptor:
Total length:           0099
Num.intf:               04
Conf.value:             01
Conf.string:            00
Attr.:                  A0
Max.pwr:                FA

Interface descriptor:
Intf.number:            00
Alt.:                   00
Endpoints:              02
Intf. Class:            FF
Intf. Subclass:         5D
Intf. Protocol:         01
Intf.string:            00
Unknown descriptor:
Length:         11
Type:           21
Contents:       0001012581140000000013010800000705

Endpoint descriptor:
Endpoint address:       81
Attr.:                  03
Max.pkt size:           0020
Polling interval:       04

Endpoint descriptor:
Endpoint address:       01
Attr.:                  03
Max.pkt size:           0020
Polling interval:       08

Interface descriptor:
Intf.number:            01
Alt.:                   00
Endpoints:              04
Intf. Class:            FF
Intf. Subclass:         5D
Intf. Protocol:         03
Intf.string:            00
Unknown descriptor:
Length:         1B
Type:           21
Contents:       000101018240010220168300000000000016030000000000000705

Endpoint descriptor:
Endpoint address:       82
Attr.:                  03
Max.pkt size:           0020
Polling interval:       02

Endpoint descriptor:
Endpoint address:       02
Attr.:                  03
Max.pkt size:           0020
Polling interval:       04

Endpoint descriptor:
Endpoint address:       83
Attr.:                  03
Max.pkt size:           0020
Polling interval:       40

Endpoint descriptor:
Endpoint address:       03
Attr.:                  03
Max.pkt size:           0020
Polling interval:       10

Interface descriptor:
Intf.number:            02
Alt.:                   00
Endpoints:              01
Intf. Class:            FF
Intf. Subclass:         5D
Intf. Protocol:         02
Intf.string:            00
Unknown descriptor:
Length:         09
Type:           21
Contents:       000101228407000705

Endpoint descriptor:
Endpoint address:       84
Attr.:                  03
Max.pkt size:           0020
Polling interval:       10

Interface descriptor:
Intf.number:            03
Alt.:                   00
Endpoints:              00
Intf. Class:            FF
Intf. Subclass:         FD
Intf. Protocol:         13
Intf.string:            04
Unknown descriptor:
Length:         06
Type:           41
Contents:       000101038000

Addr:1(0.0.1)

```