felis/USB_Host_Shield_2.0 #380: Xbox 360 hat value doesn't changing.

Kimentii, 2018-05-06:
I opened XBOXUSB example from library, set serial speed to 9600, and uploaded. Example started normally, but in one moment LeftHat value stopped changing and my Serial started get smth like "LeftHatX: 29163 LeftHatY: 16910" all the time. I cliked other buttons, but it didn't help. I tried it several times and it happened sometimes.

Soultaker, 2018-05-07:
Is this a new xbox 360 controller and is it an official one?

Kimentii, 2018-05-08:
Not new, but official. This doesn't happen with serial speed 115200. It happens on low serial speeds.

Lauszus, 2018-06-23:
It's probably spending too much time sending the data via serial. Which Arduino are you using? I would recommend using the Teensy 3.6 or similar which is much more powerful (and has emulated serial via USB) than for instance the Arduino Uno.

closed 2018-06-23 by Lauszus