felis/USB_Host_Shield_2.0 #384: Is this host shield currently available to purchase anywhere?

dniemitalo, 2018-05-24:
TKJelectronics' store is currently closed, and the link on circuitsathome is broken. I'm interested in buying the real board if it's still available somewhere.

xxxajk, 2018-05-24:
Good question... One I don't have an answer to.

xxxajk, 2018-05-24:
I just looked on TKJ and the site looks like it is up.

xxxajk, 2018-05-24:
Ahh I just noticed the note about his traveling... But I thought that was last year?? Interesting.

dsyleixa, 2018-06-20:
isn't this host shield compatible to USB_Host_Shield_2.0 ?
https://www.ebay.de/itm/USB-Host-Shield-compatible-with-UNO-Arduino-Google-Android-ADK/272676118916?hash=item3f7cc36184:g:5QcAAOSwNxda12Dd

Lauszus, 2018-06-20:
@dsyleixa it doesn't seem like the schematic is available, so I can not check it.