felis/USB_Host_Shield_2.0 #385: USBHIDJoystick compile error on ESP32

tomorrow56, 2018-05-24:
An error occurs when compiling the program "USBHIDJoystick.ino" on ESP32.
Error log is below. I want to know how to resolve this Error.
-----
C:\Users\Masao\AppData\Local\Temp\arduino_build_247149\sketch\hidjoystickrptparser.cpp: In member function 'virtual void JoystickReportParser::Parse(USBHID*, bool, uint8_t, uint8_t*)':

C:\Users\Masao\AppData\Local\Temp\arduino_build_247149\sketch\hidjoystickrptparser.cpp:55:1: internal compiler error: in dwarf2out_var_location, at dwarf2out.c:21837

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

----

Regards,

xxxajk, 2018-05-24:
Obviously the compiler is broken. Try a different version of the compiler.

chrissunny94, 2018-05-25:
Hey , can i know the pin connection between the ESP and the Usb host mini

tomorrow56, 2018-05-25:
It follows the description of readme.md
GPIO5 : SS, GPIO17 : INT, GPIO18 : SCK, GPIO19 : MISO, GPIO23 : MOSI

Regards,

Lauszus, 2018-05-26:
As @xxxajk wrote; this is indeed a compiler bug, so there is nothing we can do.

closed 2018-05-26 by Lauszus

tomorrow56, 2018-05-27:
I understood, thanks.