felis/USB_Host_Shield_2.0 #387: Arduino MEGA ADK issue

Clem-, 2018-06-11:
Hi,
I found a similar issue than [#350](https://github.com/felis/USB_Host_Shield_2.0/issues/350) where my Arduino MEGA ADK, even if powered by external supply (5V 2A), failed to connect to USB (`OSCOKIRQ failed to assert`).

I'm on Arduino 1.8.5 and working with `ArduinoBlinkLED` example.

Any clue on the cause of this failure?
Thank you.

Lauszus, 2018-06-12:
Can you sent a link to the board you are using? They might be using different pins for the INT and SS pins.

Clem-, 2018-06-12:
Hi @Lauszus, 
Thank you for your reply.

Here you'll find all the documents regarding the board I'm using : [DFRduino MEGA ADK](https://www.dfrobot.com/product-520.html).

xxxajk, 2018-06-12:
If you could, please try UHS3 located here: https://github.com/felis/UHS30
If that works, then the bug is indeed someplace in UHS2 code someplace.