felis/USB_Host_Shield_2.0 #390: Pins to be configured for have SPI working on ATmega644p

federicobriata, 2018-06-25:
Hi,
I was trying to make working the sketch USBHIDBootKbd.ino on ATmega644p and I stuck on "OSC did not start."
After some test I've noted that problem disappear when I added:
    pinMode(MOSI, OUTPUT);
    digitalWrite(SCK, HIGH);
    digitalWrite(MOSI, HIGH);
    delay( 1200 );
before calling Usb.Init().

Lauszus, 2018-06-25:
Which hardware plugin are you using? Setting the SCK and MOSI pins to HIGH is already taken care of in the SPI library: https://github.com/arduino/ArduinoCore-avr/blob/b7c607663fecc232e598f2c0acf419ceb0b7078c/libraries/SPI/src/SPI.cpp#L53-L60, so it should not be needed. The MOSI pins should not matter, as long the SCK pins is kept constant, which is handled by the hardware.

Please try my hardware plugin here: https://github.com/Lauszus/Sanguino, which should be up to date. Also make sure you are using the newest version of the Arduino IDE.

federicobriata, 2018-06-25:
Thanks for your hw plugin, I'll give a try this evening and see if this solve. Is it your plugin compatible also with atmega 644p DIP version? I'm asking because in past I had issue with other hw plugins more focused on 644p SMD that was causing floating input pins due to some Analog pin swapped, as explained [here  ](http://forum.arduino.cc/index.php?topic=151254.0).  Anyway that become solved with plugin of Leonardo Miliani because pinout is for 644p DIP https://github.com/leomil72/megax4

About SPI pins, I though too that MOSI pin should not matter but if I comment that line the issue appear again, moreover I noted that in avrpins.h there is no define MAKE_PIN(className, pin) related to 644p, so that's why I trough that it could be something missing there.

federicobriata, 2018-06-26:
Hi again,
I confirm that the issue is present also with your hw plugin and also in this case the workaroud make the issue disappear.
Moreover I confirm that if I comment MOSI pin the issue appear again.

Lauszus, 2018-09-10:
Please update to the newest version of Arduino and try again, as 1.6.12 is quite old.

federicobriata, 2018-09-19:
I confirm that with arduino-1.8.6, the issue is gone.
I tested with this changes: https://github.com/felis/USB_Host_Shield_2.0/pull/409

closed 2018-09-19 by federicobriata