felis/USB_Host_Shield_2.0 #392: USB host shield mini + Arduino Pro Micro 3.3V

igaoger, 2018-07-02:
Hi,

I checked the USB host shield mini information, and I know it should work with Arduino Pro Mini 3.3V, the problem Arduino Pro Mini 3.3V is very difficult to power it with a UBS power （5v） supply.

Now  I brought a Arduino Pro Micro 3.3V with Shield mini, but the Pro Micro pins are little bit different（and with a mini usb port）.  So I want to know how to make it work  for a Arduino Pro Micro 3.3V with shield mini? 

Which pins are used for connection for  Arduino Pro Mini 3.3V and Shield mini? 
How could I make the Shield mini work with Pro Micro 3.3V?
[http://upload.semidata.info/www.eefocus.com/ckeditor/images/Arduino_Pro_Micro_PinOut(1).jpg](url)

ernesto-xload, 2018-07-02:
Hello @igaoger just wire the SPI pins (miso, mosisck and ss) to the same labeled arduino pins (the SS just wire to digital 10 arduino pin), power (gnd and 3.3v), also if you want that the usb host have 5v power cut the VBUS jumper and wire the VBUS pad to usb 5V (Arduino Pro Micro RAW pin)

Pinouts:
Usb host shield: https://www.circuitsathome.com/wp/wp-content/uploads/2011/02/uhm11_manual.jpg
Arduino Pro Micro: https://tinkersphere.com/1707-thickbox_default/pro-micro-3-3v-8mhz-arduino-compatible-atmega32u4-breakout.jpg

Best regards

Lauszus, 2019-01-06:
Thanks @ernesto-xload for answering the question. @igaoger please remember to close issues in the future if your question has been answered :)

closed 2019-01-06 by Lauszus