felis/USB_Host_Shield_2.0 #399: Changing range of mouse *mi

trzywu, 2018-07-28:
Hello. I'm using usb shield to get information about dx and dy from my mouse, but my problem is small range of it (-127,127). I saw in lib hidboot.h  that they are uint_8, but i can't properly change them. If i change them from this 
```
struct MOUSEINFO {

        struct {
                uint8_t bmLeftButton : 1;
                uint8_t bmRightButton : 1;
                uint8_t bmMiddleButton : 1;
                uint8_t bmDummy : 5;
        };
        int8_t dX;
        int8_t dY;
};

class MouseReportParser : public HIDReportParser {

        union {
                MOUSEINFO mouseInfo;
                uint8_t bInfo[sizeof (MOUSEINFO)];
        } prevState;

```
to this below.  I see on my serial that i get first very big strange number and 0. Before i was getting 24,54 (dx,dy)  
```
struct MOUSEINFO {

        struct {
                uint8_t bmLeftButton : 1;
                uint8_t bmRightButton : 1;
                uint8_t bmMiddleButton : 1;
                uint8_t bmDummy : 5;
        };
        int16_t dX;
        int16_t dY;
};

class MouseReportParser : public HIDReportParser {

        union {
                MOUSEINFO mouseInfo;
                uint8_t bInfo[sizeof (MOUSEINFO)];
        } prevState;

public:
```

Lauszus, 2018-07-28:
That struct is only used for parsing the report. Changing it won't alter the report sent by the mouse. Your mouse probably sends out data with a higher resolution on another interface. Take a look at some of the other HID examples: https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/HID, as you will have to write your own driver for your mouse.

What is the reason you need a higher resolution anyway? What are you trying to control?

trzywu, 2018-07-28:
Well. I'm building a movement detector without contact. So I'm using a mouse sensor. But Arduino send to serial only value uint8, so I want to have a bigger range. Of course, its prototype,  and I can do it really slowly to have a precise result.  My mouse has 1200 dpi,  so i can have only 0,26 cm per detection.

Lauszus, 2018-07-28:
If you just want to have a bigger range, than you simply use the Arduino map function: https://www.arduino.cc/reference/en/language/functions/math/map/.

If you need to have a higher resolution than signed 8-bit, then you need to parse one of the other interfaces.