felis/USB_Host_Shield_2.0 #401: SPRESENSE from SONY

yokozawa766, 2018-08-11:
SONY developped new Arduino board , called SPRESENSE. I saw SPI interface on the main board. i have also seen modifies for ESP32 on this GitHub library and the library uses only SPI pins (MOSI, MISO, SCK, SS, INT) on ESP32, I mean the library doesn't use so many other GPIOs, so I guess this library can also develop modify for SPRESENSE.

By the way, I want new BT interfaces between SPRESENSE and Dualshock4 (PS4 Controller). ROHM has started selling new BLE (Bluetooth Light Energy) add-on board for SPRESENSE main board. I don't know sure whether this libray can use this add-on board. If not, the new BT add-on board between SPRESENSE and Dualshock4 can become business. ( I guess this usage of interface is in need of wireless communication, so only BT is enough, USB on-wire communication is not needed.)

xxxajk, 2018-08-12:
Are you able to send me one of these?
If so, I should be able to do support for it.
Also, please note that since the ICSP connection is missing, that some modifications will be required if using the full sized shield.

yokozawa766, 2018-08-12:
Thanks for responding. Do you mean you ban support library modification if i can send usb host shield for SPRESENSE ? Or just sending you data or link of SPRESENSE ?

closed 2018-08-12 by yokozawa766

yokozawa766, 2018-08-13:
You would pointed out that mini board doesn’t have 5V power, like Spresense main board.

We cannot implement USB master-side connector on it.

 

I guess there are two choices for that issue.

-       Making usb host shield for extention i/o board of Spresense like you have mentioned .

-       Or another way is that Bluetooth metal cap module on the mini board with SPI interface.

These metalcap modules work usually by 3.3V power.

 

Let me see…

xxxajk, 2018-08-13:
If you can send me the hardware, I can make it work, and send you something that works back.

xxxajk, 2018-08-13:
I forgot to say, that for the effort I would be keeping the hardware in order to be able to maintain the library. I would be sending a board that will work with your setup. It isn't that difficult.

yokozawa766, 2018-08-16:
I am making new pcb now like this. 

It will take one week probably.

 

 

I am also planning to send this hardware for you to work PS4BT on this board after this.

yokozawa766, 2018-08-22:
Dear Andrew,

 

Today I received printed circuit board.

I will check this board from now on.

 

 

Sincerely,

Kazunori YOKOZAWA

yokozawa766, 2018-08-30:
Finally, implemented.