felis/USB_Host_Shield_2.0 #402: USB.Init() function looping forever

asym1, 2018-08-15:
Hi all,

I'm using a Teensy 3.2 along with the USB Host Shield Mini. I installed both the USB_Host_Shield_2.0 library and spi4teensy3 library. I tried running the XBOXUSB example, as I have a wired Xbox 360 controller plugged in the Shield.

This example, aswell as the USB_desc example both get stuck on the USB.Init() line ( https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Xbox/XBOXUSB/XBOXUSB.ino#L23 )
I have also set ENABLE_UHS_DEBUGGING in settings.h to 1 ( https://github.com/felis/USB_Host_Shield_2.0/blob/master/settings.h#L42 )
and
uncommented EXTRADEBUG and PRINTREPORT in XBOXUSB.cpp ( https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXUSB.cpp#L20 )
although I'm not sure where this debugging information would be shown? I don't see it show up in the Serial Monitor in the Arduino IDE.

Any help towards resolving this issue would be greatly appreciated. Thanks in advance!

xxxajk, 2018-08-15:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/settings.h#L42

xxxajk, 2018-08-15:
> Also make sure that you have modified the mini shield to supply 5v. The
> default is 3.3v, which is going to cause problems.

See this article for all info.
https://www.circuitsathome.com/mcu/teensy-3-0-now-supported-by-the-usb-host-library/

asym1, 2018-08-15:
The VBUS jumper is cut and the pad is connected to the VIN of the Teensy 3.2.
Serial.begin() is called in the setup of the XBOXUSB example... 
So I'm not sure where the printing of the debug information is going wrong

xxxajk, 2018-08-15:
Connect to the vbus for the 5v. IIRC vin is diode protected, and an input
only. I'll check the schematic to be certain.

xxxajk, 2018-08-15:
Vin should be ok. I've always used the vbus pad. I suppose that it
shouldn't matter. Have you checked the voltage on the pin?

asym1, 2018-08-15:
https://www.pjrc.com/teensy/schematic32.gif

Vin is directly connected to Vusb on the Teensy 3.2. I checked the Vbus on the Shield and there's 5V on there aswell.

xxxajk, 2018-08-15:
Try running the boardqc to make sure that everything is ok with your wires.

asym1, 2018-08-15:
The problem is on Line 36 Usb.Init() is called before any tests are actually performed:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/board_qc/board_qc.ino#L36

So it just gets stuck on that line. Maybe I selected a wrong option in the "Tools" section of the Arduino IDE? These are my current settings:
https://gyazo.com/d82f35c2cb58a258f0d5b1634b68bcaf

gdsports, 2018-08-15:
These kinds of problems are usually wrong wiring. Post a table or diagram of your wiring.

asym1, 2018-08-15:
Hi @gdsports,

I think you just replied to my thread on the PJRC forum (https://forum.pjrc.com/threads/53452-USB-Host-Shield-with-Teensy-3-2). The thing is I have an older Teensy with a preloaded program which uses the exact same wiring and there the xbox controller works without an issue. That's why I doubt it's a wiring issue.

Anything else I might try?

xxxajk, 2018-08-16:
As I said, try the board QC program.

asym1, 2018-08-16:
@xxxajk 

> The problem is on Line 36 Usb.Init() is called before any tests are actually performed:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/board_qc/board_qc.ino#L36
So it just gets stuck on that line. Maybe I selected a wrong option in the "Tools" section of the Arduino IDE? These are my current settings:
https://gyazo.com/d82f35c2cb58a258f0d5b1634b68bcaf

As I said in an earlier reply. So the output of that program is literally the following:

> Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision

And then it gets stuck on the Usb.Init() function

xxxajk, 2018-08-16:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/settings.h#L86
try changing that to a zero and see if it works.
If not, then there is a wiring issue, perhaps even that your wires are too long.

asym1, 2018-08-17:
@xxxajk Setting that to zero actually fixed the problem! Not sure why tho but I'm glad it's resolved!

xxxajk, 2018-08-17:
Probabbly means there is a conflict with the built-in library. As far as I can remember PJRC has made the SPI faster. I'll have to look into this a bit deeper. Glad it is now working for you!

Lauszus, 2019-01-06:
@xxxajk is there any reason why ```USE_SPI4TEENSY3``` is still defined to 1 by default? A lot of improvement has happened to the SPI library including transactions etc.

wiesl, 2019-05-30:
I had the same issue, but it is a compiler optimization problem (at least in my environment):

NOT OK:
- Optimize: Faster
- CPU speed: e.g. 96MHz (overclock)

OK:
- Optimize: Fast
- CPU speed: 48 MHz, 72MHz, 96MHz (overclock)

it works then with the built-in SPI library as well as SPI4TEENSY3

BTW: you can use
`#define USE_SPI4TEENSY3 0`
to use the build-in library without changing global source code

Versions:
- Arduino: 1.8.9
- Teensyduino: 1.46
- USB Host Shield Library 2.0: 1.3.2
- spi4teensy3: Latest commit 70a2360 on Aug 14, 2017

Hardware:
- Teensy 3.2
- USB Host Shield Mini 1.1

greguz, 2019-11-02:
Same problem here, solved by setting optimize option to fast.

Thank you @wiesl!

Software:
- Arch Linux
- Arduino 1.8.5
- Teensyduino 1.44
- USB Host Shield Library 2.0: 1.3.2
- spi4teensy3: Latest commit 70a2360 on Aug 14, 2017

Hardware:
- Teensy 3.6
- Marked as *UHS mini 2.0* (Chinese clone from eBay)