felis/USB_Host_Shield_2.0 #403: Random data corruption in outTransfer

randomricky2018, 2018-08-23:
Hi,

I'm a super beginner in Arduino, usb and these drivers stuff, not sure if I understand the problem correctly. :(

I was trying to write a Android Adb host using this library, and I found that the messages that I sent sometimes (or always) corrupted. (The bytes in Usb.outTransfer does not match what's being received).

Here's my thought: (correct me if i'm wrong :) )

In UsbTransfer, due to nak bug, if nak limit is > 1, it will retry when it receives nak.
To restore it from nak, we do "regWr(rSNDFIFO, *data_p);".
However, *data_p is corrupted (the content is not the original content anymore), as in bytesWr(), we do "USB_SPI.transfer(data_p, nbytes);", which read AND write bytes in data_p by SPI.
Hence, "regWr(rSNDFIFO, *data_p);" does not write the original first byte anymore, it writes a value read from SPI.

Does it make sense or if I misunderstood sth?

Thanks a lot!

gdsports, 2019-06-28:
I have seen this problem sending to a slow printer which generates lots of NAKs. The root cause is the SPI.transfer function reads AND writes the data buffer so in Usb.cpp|USB::OutTransfer the following line sends garbage instead of the correct value.

https://github.com/felis/USB_Host_Shield_2.0/blob/49421b6ce40442d5afe3097ab0f3e65b8fc69aa5/Usb.cpp#L376

bytesWr() calls SPI.transfer() which sends data to the USB out FIFO but also reads from the SPI bus thus over writing the original values at data_p.

https://github.com/felis/USB_Host_Shield_2.0/blob/49421b6ce40442d5afe3097ab0f3e65b8fc69aa5/Usb.cpp#L337

My fix is to save the first byte of data_p before bytesWr() then use it for NAK retry, if needed.

```
		uint8_t save_byte = *data_p;
                bytesWr(rSNDFIFO, bytes_tosend, data_p); //filling output FIFO

                regWr(rSNDFIFO, save_byte);

```
https://github.com/felis/USB_Host_Shield_2.0/issues/403 might also be related.

Lauszus, 2019-08-05:
Fixed in #473

closed 2019-08-05 by Lauszus