felis/USB_Host_Shield_2.0 #406: Arduino MKR1000 support

jigokutenshi, 2018-08-27:
Hello,

I had a perfect project planned using the the Arduino MKR1000 board and connecting a ps4 controller via bluetooth. Unfortunately, I get the "Please define board in avrpins.h" error. Is it possible you can add support for this type of board?

Lauszus, 2019-01-06:
@jigokutenshi is this still relevant? If you ship me one, then I'll add support for it.

jigokutenshi, 2019-01-09:
I can! Just let me know what steps to follow and I'll ship you the one I have.

Lauszus, 2019-01-10:
Thanks, please email me at lauszus@gmail.com :)

kizmit99, 2019-06-02:
I'm curious whether any progress has been made on this?
I'm trying to use this library with a MKRZero to control (a clone of) the mini-shield as a prototype for a custom design using the SAMD21 and MAX3421E.
I'm willing to fork and take my own cut at modifying avrpins.h, but if some progress has already been achieved, I'd prefer to not reinvent the wheel...
Thanks/

kizmit99, 2019-06-09:
Just to follow-up -- I was successful in getting this library to work on the MKRZero board, but did run into a few issues that make the port significantly less than pretty...  (so definitely not a good candidate for a merge request).  Documenting here in case anyone in future tries to make a similar port:
* avrpins.h does need to be updated, the following is working for me:
```
#endif // Arduino pin definitions

#elif defined(__arm__)

// pointers are 32 bits on ARM
#define pgm_read_pointer(p) pgm_read_dword(p)

#if defined(ARDUINO_SAMD_MKRZERO)

#define MAKE_PIN(className, pin) \
class className { \
public: \
  static void Set() { \
    digitalWrite(pin, HIGH); \
  } \
  static void Clear() { \
    digitalWrite(pin, LOW); \
  } \
  static void SetDirRead() { \
      pinMode(pin, INPUT); \
  } \
  static void SetDirWrite() { \
      pinMode(pin, OUTPUT); \
  } \
  static uint8_t IsSet() { \
    return digitalRead(pin); \
  } \
};

MAKE_PIN(P0, 0);
MAKE_PIN(P1, 1);
MAKE_PIN(P2, 2);
MAKE_PIN(P3, 3);
MAKE_PIN(P4, 4);
MAKE_PIN(P5, 5);
MAKE_PIN(P6, 6);
MAKE_PIN(P7, 7);
MAKE_PIN(P8, 8);
MAKE_PIN(P9, 9);
MAKE_PIN(P10, 10);
MAKE_PIN(P11, 11);
MAKE_PIN(P12, 12);
MAKE_PIN(P13, 13);
MAKE_PIN(P14, 14);
MAKE_PIN(P15, 15);
MAKE_PIN(P16, 16);
MAKE_PIN(P17, 17);
MAKE_PIN(P18, 18);
MAKE_PIN(P19, 19);
MAKE_PIN(P20, 20);
MAKE_PIN(P21, 21);
MAKE_PIN(P22, 22);
MAKE_PIN(P23, 23);
MAKE_PIN(P24, 24);
MAKE_PIN(P25, 25);
MAKE_PIN(P26, 26);
MAKE_PIN(P27, 27);
MAKE_PIN(P28, 28);
MAKE_PIN(P29, 29);
MAKE_PIN(P30, 30);
MAKE_PIN(P31, 31);
MAKE_PIN(P32, 32);
MAKE_PIN(P33, 33);
MAKE_PIN(P34, 34);
MAKE_PIN(P35, 35);

#undef MAKE_PIN

#elif defined(CORE_TEENSY) && (defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__))
```

* The MKRZero definitions for the PIN_SPI_MOSI (etc) have a "u" in them that the macros used in usbhost.h do not like.  I made the following change in usbhost.h:
```
/* SPI pin definitions. see avrpins.h   */
#if defined(ARDUINO_SAMD_MKRZERO)
typedef SPi< P9, P8, P10, P4 > spi;
#elif defined(PIN_SPI_SCK) && defined(PIN_SPI_MOSI) && defined(PIN_SPI_MISO) && defined(PIN_SPI_SS)
```

* This is the biggy - there is a name conflict between USB and UsbDevice as defined in the UHS2 libs and the native USB libs for MKRZero.  I had to change the name of USB to USBMAX and UsbDevice to UsbDeviceMax throughout the library.

With those changes I've got this library and the examples (at least the ones I tried) working on a MKRZero board.
Thanks/

gdsports, 2019-06-14:
An alternative is to use the SAMD21 USB controller in host mode. No max3421 chip needed. Not all of the drivers work so there are trade offs.

https://github.com/gdsports/USB_Host_Library_SAMD