felis/USB_Host_Shield_2.0 #408: USB host for SAMD

gdsports, 2018-09-11:
I ported over some drivers (ftdi, pl2303, cdcacm, HID, MIDI, Xbox, and PS) to the Arduino Zero, M0, and MKR family. The project uses the built-in USB OTG hardware also known as the native USB port. The MAX3421E controller is not used. Changes are required to the Arduino SAMD board package so this is best for experienced developers.

At least for now, I am not planning to submit a pull request because it means littering the code with conditionals. However, I am open to suggestions.

https://github.com/gdsports/USB_Host_Library_SAMD

gdsports, 2018-09-12:
Adding SAMD USB host support to UHS3 looks like a much bigger job than porting the UHS2 drivers to the SAMD USB host stack. I will look at UHS3 after I get a bit more working on my project.

xxxajk, 2018-09-12:
Actually... it is much easier than you might think. 

All you need to do is 

1. Have a 1ms timer, which is usually provided by the USB hardware by generating frame interrupts.
2. Talk to the USB hardware in the ISR
3. Catch connecting, disconnecting
4. TX/RX packets

Once done, the drivers will just work.

There are already examples for USB Host Shield and Kinetis low/full speed. 

I still need to work out Kinetis high speed for the full USB 2.0 on the MK6x chips.
That I will be resuming work on next week.

Lauszus, 2019-01-06:
Closing this, as this will eventually be supporting in UHS3.

closed 2019-01-06 by Lauszus