felis/USB_Host_Shield_2.0 #410: [Question] Need help debugging INVALID_CSW error

marcio-ao, 2018-09-19:
Hello,

I am trying to troubleshoot a problem I are seeing with certain flash drives on our printer prototypes. I are noticing a peculiar situation where if I power on the printer with a USB drive inserted, it is able to read files just fine. But, if you insert it afterwards, it fails the initialization. This happens only on some USB drives, and we've used others without problems.

I turned on USB logging and the cause of the failure seems to be the "Invalid CSW" error. Interestingly, when I edit "massstorage.cpp" and skip the check (i.e IGNORE_INVALID_CSW=1), drive which would otherwise fail work just fine.

```
uint8_t BulkOnly::Transaction(...) {
   ...
                if(usberr == hrSUCCESS) {
                        if(IsValidCSW(&csw, pcbw)) {
                                ...
                        } else {
                                // NOTE! Sometimes this is caused by the reported residue being wrong.
                                // Get a different device. It isn't compliant, and should have never passed Q&A.
                                // I own one... 05e3:0701 Genesys Logic, Inc. USB 2.0 IDE Adapter.
                                // Other devices that exhibit this behavior exist in the wild too.
                                // Be sure to check quirks in the Linux source code before reporting a bug. --xxxajk
                                Notify(PSTR("Invalid CSW\r\n"), 0x80);
                                #if IGNORE_INVALID_CSW == 1
                                        return MASS_ERR_SUCCESS;
                                #endif
                                ResetRecovery();
                                return MASS_ERR_INVALID_CSW;
                        }
   ...
}
```

Given the code comment, I would chalk this up as being a non-standards compliant drive, except that it works just fine if the printer is powered on *with* the drive inserted. If it was a standards compliance issue, it wouldn't behave differently in both cases.

So, two questions:

1) Any thoughts at to why this is happening?
2) How dangerous is it to skip the INVALID_CSW check? I worry that by skipping it I may simply be masking another problem.

I know nothing about USB protocols, so anything that can get me moving in the right direction would be very helpful!

Thank you!

-- Marcio

----

USB Logs:

```
MS ConfigureDevice
MS Init
Addr:01
NC:01
Conf.Val: 01
Iface Num: 00
Alt.Set: 00
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Conf.Val: 01
Iface Num: 00
Alt.Set: 00
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Conf:01
MaxLUN: 00

Inquiry
---------
CBW.dCBWTag: 00000001
CBW:		OK
Data Stage:	OK
CSW:		OK

TestUnitReady
-----------------
CBW.dCBWTag: 00000002
CBW:		OK
Data Stage:	OK
CSW:		OK

SCSI Error: 01
LUN: 00

RequestSense
----------------
CBW.dCBWTag: 00000003
USB Error: 04
Index: 01
============================ DAT: 04

ResetRecovery
-----------------

ResetRecovery
-----------------

TestUnitReady
-----------------
CBW.dCBWTag: 00000004
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000005
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000006
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000007
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000008
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000009
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000000A
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000000B
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000000C
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000000D
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000000E
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000000F
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000010
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000011
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000012
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000013
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00
MS configured

Drive inserted

LUN zero is not good

echo:SD init fail

TestUnitReady
-----------------
CBW.dCBWTag: 00000014
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000015
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000016
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000017
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000018
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 00000019
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000001A
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000001B
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000001C
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000001D
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

TestUnitReady
-----------------
CBW.dCBWTag: 0000001E
CBW:		OK
Data Stage:	OK
CSW:Sig error
Invalid CSW

ResetRecovery
-----------------
Gen SCSI Err: 07
LUN: 00

```

marcio-ao, 2018-09-20:
Okay, we added a few print functions to IsValidCSW and found a single bit error in the returned value:

TestUnitReady
-----------------
CBW.dCBWTag: 00000013
CBW:		OK
Data Stage:	OK
CSW:Sig error
pcsw->dCSWSignature: 0x43425355
MASS_CSW_SIGNATURE: 0x53425355
Invalid CSW

It looks like reducing the clock rate on our SPI interface has cleared the error. While it is a bit peculiar that it is only affecting the first bit consistently, at least I have a workaround.

xxxajk, 2018-09-21:
Not too surprised.
There are things that you can do to get higher speeds.
Try to make the clock and data tracks are the same length. This will help with clock v.s. data race conditions.
If you are level shifting the signals, it can be helpful to use chips that are as fast as you can obtain, or, to use a pull-up resistor and a diode. Either of these will remove latency issues a lot! If not and everything is 3.3v, then you might be seeing distortion on the signals.
Sometimes you may need to add either pull up or pull down resistors. Check your signalling with a scope, make sure it is clean.

marcio-ao, 2018-09-21:
@xxxajk: We are actually running this via a ribbon cable of about six inches and it the cable is daisy chained over at least three devices. We scoped it up and saw that it was being driven at 1.6MHz -- diving it at ~180 kHz seems to work much better.

Will close this ticket.

closed 2018-09-21 by marcio-ao